ev3dev/ev3dev #1192: New snapshot

dlech, 2018-11-13:
### What are you trying to do?

Get some testers to test the 2018-11-13 snapshot SD card images. I would like to make this the main download on the main downloads page if we don't find any serious issues.

### What did you expect to happen?

People should make sure it boot, make sure network connections work. run some motors, etc. Leave a comment here saying what you tested.

### What actually happened?

We shall see... :smiley: 

### What hardware and software are you using (including version numbers)?

https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-11-13/

shelbydavis, 2018-11-14:
Downloaded the SD image, booted up on an EV3 with a rechargeable battery.
Connected to wifi via Edimax USB dongle, entering password via brickman interface.
Verified that the light sensor works via brickman interface, and ran a quick python script running one motor, both through the MSFT Visual editor (on a mac) and via SSH on the command line.

JorgePe, 2018-11-18:
fresh SD card, EV3 with LiPo, Edimax USB dongle and CSR 4.0 BLE dongle (and mini USB hub).
Bluetooth ON, BLE works through hci1 as expected.
Wi-Fi ON, connected to my Android hotspot and also to my job's wi-fi. SSH works (through wi-fi).
Ultrasonic sensor works in default mode.
Touch sensor works.
Battery levels read OK.
Cable adapter to RCX motor works setting OutA to dcmotor in brickman interface.
Just plain shell commands and bash scripts, no python yet.

dlech, 2018-11-20:
Thank you @shelbydavis and @JorgePe for taking the time to test and report your findings. I've published this image as the main download on ev3dev.org now.

Since only a couple people actually took the time to respond to this issue within one week, I would like give you an ev3dev.org laptop sticker for being awesome. Please get in touch with me to claim your sticker.

closed 2018-11-20 by dlech

robl0377, 2018-11-23:
The day before yesterday I got my 'official' Wifi dongle (Netgear WN1100) which works very well on the original EV3 brick firmware (1.09E) and so I tried the new 2018-11-13 snapshot on a new class 10 32G Micro SDHC card. Used the balenaEtcher 1.4.7 to flash the image on the card:

Bootup the SD card on the EV3 brick was working perfectly.

However, when I finished entering the Wifi PSK in the Brickman menu, there was an error message, saying there is no network, although I selected my home network SSID just before.
Maybe there was some timeout expiring, because my PSK has 20 characters (upper and lower case letters and digits, no special characters). BTW: My Wifi Router only allows WPA2 with AES encryption.
And the LCD keyboard does not help entering that long PSK (maybe have a look at the LeJOS LCD keyboard: it has letters and digits on the same screen)

So I was resorting to the good old USB cable according to (https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/). This was working fine. (I got the wonderful EV3 Brick icon from the installed Programmable Brick Icons in my Windows 10 "Devices and Printers" box. 
But no connection!!!
What happened: The Windows RNDI Device got an IP 192.168.137.1, but the EV3 Brick still resorted to some Windows default 169.254.xxx.xxx. Several reboots of the EV3 and the Windows Laptop.. Tried to set DHCP in the Brickman menu but to no avail. Obviously there was no DHCP working for the Windows Device.
After explicitly setting the IP, Mask and Gateway in the Brickman menu to fit to 192.168.137.1, the USB connection was established. Hooray! 
BTW: Entering IP, Mask and Gateway could be easier if you just show a digit and dot keyboard on the LCD.
Then I could successfully follow the Setting Up Wi-Fi Using the Command Line Tutorial (https://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/).
So that is my status now.

I fear the way I did this is not the easiest/fastest way to do for beginners of robotics, so it would be great if entering the Wifi PSK in Brickman was working better. 
The other thing with the missing DHCP in Windows 10 is another story. Maybe I did not dig deep enough to find a Windows solution..

jabrena, 2018-11-24:
Hi @dlech,

I finished some full regressions for:

- BrickPi+
- BrickPi3
- PiStorms V2

Using different Raspberry Pi Boards:

- RaspberryPi 2
- RaspberryPi 3
- RaspberryPi 3 Model B+

I found the following issues in:

- BrickPi+: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/595
- PiStorms V2: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/595

I grouped by Platform, tell me if you have some doubt. In the case of PiStormsV2, the results was pretty bad.

Juan Antonio