ev3dev/ev3dev #1193: Please put latest ptvsd version in the next build

omartin2010, 2018-11-16:
### What are you trying to do?
Get the latest PTVSD version included in the next build.

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

dlech, 2018-11-20:
`ptvsd_4.2.0-1ev3dev1` is now available in the ev3dev package archive.

closed 2018-11-20 by dlech

omartin2010, 2018-11-20:
hey David, is it yet in the docker image? And in the latest ev3dev stretch image (I believe 2018-11-13)?

dlech, 2018-11-20:
nope and nope

omartin2010, 2018-11-20:
so pardon my ignorance, but how do I get it other than manually installing it? Is it just pip install ptvsd_4.2.0-1ev3dev1 ?

dlech, 2018-11-20:
sudo apt update
    sudo apt install python3-ptvsd

(leave out the 3 if are using python2)

omartin2010, 2018-11-22:
so now that worked on the BrickPI and now that I'm trying this on the lego brick, I get this :  
```
robot@ev3dev:~$ sudo apt install python3-ptvsd
[sudo] password for robot:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  python3-ptvsd
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/876 kB of archives.
After this operation, 6,356 kB of additional disk space will be used.
Selecting previously unselected package python3-ptvsd.
(Reading database ... 34529 files and directories currently installed.)
Preparing to unpack .../python3-ptvsd_4.2.0-1ev3dev1_all.deb ...
Unpacking python3-ptvsd (4.2.0-1ev3dev1) ...
Setting up python3-ptvsd (4.2.0-1ev3dev1) ...
  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_SimpleXMLRPCServer.py", line 263
    except Fault, fault:
                ^
SyntaxError: invalid syntax

``` 
and so long... any clue? I tried a few things, ran a full apt update and upgrade (which took quite some time, went out of battery - my bad - and then restarted and had that failed... not sure if it has anything to do with going out of battery (not sure if the shutdown was "clean" or not as I wasn't paying attention when it happened...).)

dlech, 2018-11-22:
It looks like python2-only code made it into the python3 package. It looks like some paths have changed from here: https://github.com/ev3dev/ptvsd/commit/ebf6f0b0073a81fb47d4ee1f7dd5d24fbfb848fe. These just need to be update to the new paths and the package rebuilt.