ev3dev/ev3dev #1194: TL-WN725N wifi dongle can't work

wangcray, 2018-11-16:
### What are you trying to do?

get TL-WN725N work

### What actually happened?

In EV3 "WI-FI", it shows: "Not available"

There's no /dev/wlan0 or sth else alike, and there's nothing related in `ifconfig`.
I did `modprobe 8188eu`. It succeeds but nothings happens.
I just googled a lot and tried much according to some methods in this issues list, but it just doesn't work.
So frustrated

### What hardware and software are you using (including version numbers)?

Here is some info:

```
# lsusb
Bus 001 Device 039: ID 0bda:1a2b Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

# uname -ra
Linux ev3dev 4.14.61-ev3dev-2.2.2-ev3 #1 PREEMPT Mon Aug 6 14:22:31 CDT 2018 armv5tejl GNU/Linux

# ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3  bytes 300 (300.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3  bytes 300 (300.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet6 fe80::16:53ff:fe62:5eba  prefixlen 64  scopeid 0x20<link>
        ether 02:16:53:62:5e:ba  txqueuelen 1000  (Ethernet)
        RX packets 462  bytes 56985 (55.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 298  bytes 46924 (45.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb1: flags=-28669<UP,BROADCAST,MULTICAST,DYNAMIC>  mtu 1500
        ether 22:16:53:62:5e:ba  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

# lsmod

Module                  Size  Used by
8188eu                724451  0
iptable_nat             1728  0
nf_conntrack_ipv4       5927  1
nf_defrag_ipv4          1634  1 nf_conntrack_ipv4
nf_nat_ipv4             5140  1 iptable_nat
nf_nat                 14692  1 nf_nat_ipv4
nf_conntrack           87388  3 nf_conntrack_ipv4,nf_nat_ipv4,nf_nat
libcrc32c               1100  2 nf_conntrack,nf_nat
usb_f_rndis            15528  2
usb_f_ecm               6278  2
u_ether                12402  2 usb_f_ecm,usb_f_rndis
ev3_pru                 4044  0
virtio_rpmsg_bus        9194  0
rpmsg_core              7085  2 ev3_pru,virtio_rpmsg_bus
ti_pru_rproc            3036  0
remoteproc             21570  1 ti_pru_rproc
virtio                  7203  2 virtio_rpmsg_bus,remoteproc
virtio_ring            15777  2 virtio_rpmsg_bus,remoteproc
ti_pru_intc             4496  0
ti_pruss                1426  0
suart_emu              21593  0
uinput                  8046  0
libcomposite           41646  16 usb_f_ecm,usb_f_rndis
sunrpc                228280  1
ip_tables              10235  1 iptable_nat
x_tables               19456  1 ip_tables
sr_mod                 13014  0
cdrom                  28732  1 sr_mod

```

dlech, 2018-11-16:
It doesn't look like this device has mainline Linux support. e.g. https://ubuntuforums.org/showthread.php?t=2389602&page=3

So, you can try to compile your own driver or get a different Wi-Fi dongle. https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles

Jaif, 2018-11-16:
If this is a TL-WN725N **v2** then you can force the 8188eu driver to recognize the device by issuing this command after you have loaded the 8188 driver:
`sudo echo 0bda 1a2b > /sys/bus/usb/drivers/r8188eu/new_id`

If that works, you can make a new udev rule to fix it automaticaly.
`nano /etc/udev/rules.d/99-TLWN725Nv2.rules`
Add the following line:
`ACTION=="add", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="1a2b", RUN+="/sbin/modprobe 8188eu" RUN+="/bin/sh -c 'echo 0bda 1a2b > /sys/bus/usb/drivers/r8188eu/new_id'"`

If it is a TL-WN725N **v1** then you need a different driver.

wangcray, 2018-11-17:
Thank you for your help.
Starting to try...
I'll renew it when getting proceeded

wangcray, 2018-11-18:
get sth but not get to the final.

Based on the combined info from @dlech and @Jaif
```
/* change the dongle mode from storage into wifi */
# usb_modeswitch -KW -v 0bda -p 1a2b

# lsusb
Bus 001 Device 039: ID 0bda:b711 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ohci-da8xx/1p, 12M
    |__ Port 1: Dev 115, If 0, Class=Vendor Specific Class, Driver=, 12M

# echo 0bda b711 | tee > /sys/bus/usb/drivers/r8188eu/new_id

# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ohci-da8xx/1p, 12M
    |__ Port 1: Dev 39, If 0, Class=Vendor Specific Class, Driver=r8188eu, 12M
```

It seems the driver recognizes the dongle now. In EV3 "WI-FI", there's "Powered" checkbox showed up. After check it, there's a "Start Scan" selection.

However, if select the "Start Scan", it says "Error GDBus.Error:net.connman.Error.NoCarrier:No carrier".

In ssh command line, what I try and get:

```
# connmanctl
connmanctl> enable wifi
Error wifi: Operation not permitted
```

There's still no /dev/wlan0 and nothing about wifi from 'ifconfig'.
I post `dmesg` info below. I know little about linux kernel tech. But there's sth like "r8188eu 1-1:1.0 wlx00e04c818802: renamed from wlan0"

Any good idea?

```
[  415.543150] print_req_error: I/O error, dev sr0, sector 4096
[  415.543194] Buffer I/O error on dev sr0, logical block 512, async page read
[  415.673042] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  415.673134] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 ff fa 00 00 02 00
[  415.673182] print_req_error: I/O error, dev sr0, sector 262120
[  415.673228] Buffer I/O error on dev sr0, logical block 32765, async page read
[  416.242971] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  416.243067] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 ff fc 00 00 02 00
[  416.243114] print_req_error: I/O error, dev sr0, sector 262128
[  416.373008] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  416.373104] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 ff fc 00 00 02 00
[  416.373153] print_req_error: I/O error, dev sr0, sector 262128
[  416.373202] Buffer I/O error on dev sr0, logical block 32766, async page read
[  417.092930] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  417.093025] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 fe 80 00 00 3c 00
[  417.093073] print_req_error: I/O error, dev sr0, sector 260608
[  417.222891] Buffer I/O error on dev sr0, logical block 32576, async page read
[  417.482885] Buffer I/O error on dev sr0, logical block 512, async page read
[  417.612868] Buffer I/O error on dev sr0, logical block 32765, async page read
[  419.633589] usb 1-1: USB disconnect, device number 52
[  420.167662] usb 1-1: new full-speed USB device number 53 using ohci-da8xx
[  420.438621] usb 1-1: New USB device found, idVendor=0bda, idProduct=1a2b
[  420.438687] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  420.438726] usb 1-1: Product: DISK
[  420.438763] usb 1-1: Manufacturer: Realtek
[  420.466035] usb-storage 1-1:1.0: USB Mass Storage device detected
[  420.478293] scsi host0: usb-storage 1-1:1.0
[  421.505426] scsi 0:0:0:0: CD-ROM            Realtek  USB Disk autorun 1.00 PQ: 0 ANSI: 0 CCS
[  421.531941] sr 0:0:0:0: [sr0] scsi-1 drive
[  421.534501] sr 0:0:0:0: Attached scsi CD-ROM sr0
[... 282 lines cut ...]
[  463.598727] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  463.598820] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 fe 80 00 00 02 00
[  463.598869] print_req_error: I/O error, dev sr0, sector 260608
[  463.598918] Buffer I/O error on dev sr0, logical block 32576, async page read
[  463.738631] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  463.738726] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 04 00 00 00 02 00
[  463.738774] print_req_error: I/O error, dev sr0, sector 4096
[  463.868665] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[  463.868759] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 00 04 00 00 00 02 00
[  463.868805] print_req_error: I/O error, dev sr0, sector 4096
[  463.868849] Buffer I/O error on dev sr0, logical block 512, async page read
[  464.008046] usb 1-1: USB disconnect, device number 58
[  464.433975] usb 1-1: new full-speed USB device number 59 using ohci-da8xx
[  464.706446] usb 1-1: New USB device found, idVendor=0bda, idProduct=b711
[  464.706518] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  464.706564] usb 1-1: Product: 802.11n WLAN Adapter
[  464.706688] usb 1-1: Manufacturer: Realtek
[  464.706730] usb 1-1: SerialNumber: 00E04CB82101
[  587.056564] 8188eu: loading out-of-tree module taints kernel.
[  587.136212] usbcore: registered new interface driver r8188eu
[  591.651721] Chip Version Info: CHIP_8188E_Normal_Chip_UMC_UNKNOWN_CUT(7)_1T1R_RomVer(0)
[  591.651801] EEPROM ID(0xffff) is invalid!!
[  591.651828] EEPROM ID = 0xffff
[  592.565396] r8188eu 1-1:1.0 wlx00e04c818802: renamed from wlan0
```

omartin2010, 2018-11-22:
you might need to look around wpasupplicant (/etc/wpasupplicant/wpa-supplicant-wlx......conf) or the default wpa-supplicant.conf should contain the required config (ssid, wpa key, ...). Try using the wpa cli as well to scan... and ip link should show that wlx device (which is the new wlan device's name).

wangcray, 2018-11-25:
@omartin2010 thank you for your response. 
But I don't think it will work when there's no interface (e.g., wlan0).

Ergus, 2020-02-25:
Hi:
Sorry for necropumping this. After more than a year. Did anyone found a way to make 0bda:b711 work?

wangcray, 2020-02-26:
sadly no...

AlexanderMakarov, 2020-07-21:
I guess it is popular WiFi dongle. Still no luck? Only Windows?