ev3dev/ev3dev #1198: dpkg: error processing package python3-ptvsd

JorgePe, 2018-11-22:
using November 2018 stretch snapshot image

`Linux ev3dev 4.14.80-ev3dev-2.3.1-ev3 #1 PREEMPT Tue Nov 13 13:08:11 CST 2018 armv5tejl GNU/Linux`

'sudo apt update' reported a new package, 'python3-ptvsd' so upgrade... but fails

lots of messages `SyntaxError: Missing parentheses in call to 'print' `

JorgePe, 2018-11-22:
oops, related to issue #1193

dlech, 2018-11-23:
Just pushed out `ptvsd_4.2.0-1ev3dev2` with a possible fix, so please try again.

cc: @omartin2010

omartin2010, 2018-11-23:
it works, @dlech : thanks!

closed 2018-11-23 by dlech

salamanders, 2019-11-24:
Not sure if this is the same issue: Same package, similar errors.
BrickPi3, raspberrypi, buster.

```
robot@ev3dev:~$ sudo apt install linux-image-ev3dev-rpi2
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-image-ev3dev-rpi2 is already the newest version (34).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up python3-ptvsd (5.0.0~a5-1ev3dev1) ...
  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_SimpleXMLRPCServer.py", line 263
    except Fault, fault:
                ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_SocketServer.py", line 339
    print '-'*40
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('-'*40)?

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_inspect.py", line 293
    raise TypeError, 'arg is a built-in module'
                   ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_pkgutil_old.py", line 543
    except IOError, msg:
                  ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_xmlrpclib.py", line 294
    True, False = True, False
    ^
SyntaxError: can't assign to keyword

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydevd_bundle/pydevconsole_code_for_ironpython.py", line 105
    except SyntaxError, err:
                      ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydevd_bundle/pydevd_exec.py", line 3
    exec exp in global_vars, local_vars
           ^
SyntaxError: Missing parentheses in call to 'exec'

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydev_ipython/inputhookglut.py", line 98
    print '\nKeyboardInterrupt'
                              ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('\nKeyboardInterrupt')?

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/_check.py", line 2
    print add_code_to_python_process.run_python_code(3736, "print(20)", connect_debugger_tracing=False)
                                   ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/winappdbg/plugins/do_example.py", line 37
    print "This is an example command."
                                      ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("This is an example command.")?

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/winappdbg/plugins/do_exchain.py", line 40
    print "Exception handlers for thread %d" % thread.get_tid()
                                           ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/winappdbg/plugins/do_exploitable.py", line 46
    print "-" * 79
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("-" * 79)?

dpkg: error processing package python3-ptvsd (--configure):
 installed python3-ptvsd package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 python3-ptvsd
E: Sub-process /usr/bin/dpkg returned an error code (1)
robot@ev3dev:~$

dlech, 2019-11-24:
If an issue is closed, it is always best to open an new issue unless you are 100% sure the problem is not really fixed.