ev3dev/ev3dev #1199: ev3dev Python very slow

wlklch, 2018-12-03:
running I python program either directly from lego brick, or from my PC through SSH and USB takes a minute to load before program really starts work

This is described in closed issue https://github.com/ev3dev/ev3dev/issues/1022 

however, I tried several of the ideas there - even several hours of sudo apt upgrade - without any success. Furthermore, it is not clear to me, how the issue really got solve to close the issue!

Any help?

WasabiFan, 2018-12-05:
We need some more information to understand what's going on.

First, please include an answer to the following (from our issue template):

```
### What hardware and software are you using (including version numbers)?

<!-- Tip: copy and paste the output of `ev3dev-sysinfo -m` here -->
```

Second, what is the output of `dpkg-query -l python3-ev3dev*`? Finally, could you show us a code sample (in particular, the top of the file where you import the library)?

wlklch, 2018-12-06:
Here are the desired infos. I simplified the script a bit, so its a bit faster. 

- The first run after boot is more than 3 times faster than the following one (still tedious, but not catastrophic).
- Python module cache seems to work 
- top shows that python runs the whole time at +-40% CPU, furthermore there are two processes kworker*/u* that together use also nearly 40%cpu, even a long time after python has finished

**Infos from 6.12.18
ev3dev-sysinfo -m**

```
    <!-- Copy everything between these lines -->
    **System info (from `ev3dev-sysinfo`)**
    ```
    Image file:         ev3dev-stretch-ev3-generic-2018-08-06
    Kernel version:     4.14.80-ev3dev-2.3.1-ev3
    Brickman:           0.10.0
    BogoMIPS:           148.88
    Board:              board0
    BOARD_INFO_HW_REV=7
    BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
    BOARD_INFO_ROM_REV=6
    BOARD_INFO_SERIAL_NUM=001653535C31
    BOARD_INFO_TYPE=main
    ```
    <!-- Copy everything between these lines -->
```
**dpkg-query -l python3-ev3dev***   

```
    Desired=Unknown/Install/Remove/Purge/Hold
    | Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
    |/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
    ||/ Name                       Version            Architecture       Description
    +++-==========================-==================-==================-=========================================================
    ii  python3-ev3dev             1.2.0              all                Python language bindings for ev3dev
    ii  python3-ev3dev2            2.0.0~beta2        all                Python language bindings for ev3dev

```    
**python programm ir1.py**

```
    #!/usr/bin/env python3

    from datetime import datetime
    print(str(datetime.now()), "00 import datetime", __file__)
    import time
    import threading
    from queue import Queue, Empty
    import sys

    print(str(datetime.now()), "01 import sys", __file__)

    #from modmod import *

    #print(str(datetime.now()), "02 import modmod", __file__)

    from ev3dev2.motor import *
    print(str(datetime.now()), "03 import motor", __file__)
    from ev3dev2.led import Leds
    from ev3dev2.button import *
    from ev3dev2.sensor.lego import InfraredSensor
    #from ev3dev2._platform.ev3 import *
    print(str(datetime.now()), "10 imported all", __file__)

    # print('EVDEV_DEVICE_NAME', EVDEV_DEVICE_NAME)
    # for name, mod in sys.modules.items():
    #    print(name, '-->', mod)
    class IRButThr(threading.Thread):
        """ thread ev3 brick buttons """
        def __init__(self):
[... 102 lines cut ...]
        ./ir 4 end Thu Dec  6 07:44:04 UTC 2018
        Thu Dec  6 07:44:04 UTC 2018
```
        **output**

```
        ./ir 2 begin Thu Dec  6 07:43:19 UTC 2018
        2018-12-06 07:43:31.049553 00 import datetime /home/robot/ir1.py
        2018-12-06 07:43:37.800533 01 import sys /home/robot/ir1.py
        2018-12-06 07:43:42.356409 03 import motor /home/robot/ir1.py
        2018-12-06 07:44:03.174098 10 imported all /home/robot/ir1.py
        2018-12-06 07:44:03.188040 12 after class /home/robot/ir1.py
        sys.version 3.5.3 (default, Sep 27 2018, 17:25:39) 
        [GCC 6.3.0 20170516]
        2018-12-06 07:44:03.199930 platform ev3
        InfraredSensor ev3-ports:in1 
        	modes=['IR-PROX', 'IR-SEEK', 'IR-REMOTE', 'IR-REM-A', 'IR-S-ALT', 'IR-CAL'] 
        	_BUTTONS=('top_left', 'bottom_left', 'top_right', 'bottom_right', 'beacon') 
        	 commands=[]
        2018-12-06 07:44:03.515667 20 run begin /home/robot/ir1.py
        2018-12-06 07:44:03.549832 ir Heading Distance ch1 (0, None)
        2018-12-06 07:44:03.553457 19 end /home/robot/ir1.py
        ./ir 3 end Thu Dec  6 07:44:04 UTC 2018

```

WasabiFan, 2018-12-14:
The timing on the first run is about what I would expect -- importing everything takes about 8 seconds, which is slow but not particularly surprising for the EV3. On the second run it looks like everything is slower (both built-in imports and `ev3dev2`) so I'm guessing that something is chewing CPU rather than the Python library itself behaving differently. Just before you run for the second or third time, what is using CPU? Is your script still alive and using cycles, perhaps?

wlklch, 2018-12-14:
As mentioned before, we have kworker/* using over 30% cpu after the first run. And I still would like to get an answer what they are doing, or how I can find it out. I only used python and I'm not aware, that I fed any work to kworkers.

You can see that, in points 1 - 5 below - during the python runs top was not active here. It Seems that idle going back from 90% to 30% explains the nearly 4 times longer runtime.

However, running top during python, shows a different picture. Maximal CPU usage for Python decreased from 70% to 40% only, furthermore, IO-Wait decreased from a peak of 6% to permanently 0.0. But if you look at Time+ in top, you see that the total cpu consumption of Python doubled (see points 6 and 7 below) 

find the mentioned screen shots in [pythonSlow181214.pdf](https://github.com/ev3dev/ev3dev/files/2679545/pythonSlow181214.pdf)

dlech, 2018-12-14:
FYI, `kworker/*` are kernel threads. I don't know of a way to figure out exactly what each one is for.

I also wonder how much the microSD card performance affects the startup time. ([Benchmarks for comparison](https://github.com/ev3dev/ev3dev/wiki/Selecting-a-microSD-card))

wlklch, 2018-12-17:
using while true ; do { { date; cat /proc/interrupts ; } >>int.txt ; sleep 10 ; } done we see how how the interrupt counters increase. The major difference before and after the python script

```
             interrupts/sec
          boot     run    prox   
 53:         0    5240      53   cp_intc  53 Edge      ttyS1
 56:         1     642       1   cp_intc  56 Edge      spi_davinci.1

```
There are 3 columns with measurements
- boot: after boot
- run: after python was run, which is using Infraredsensor and leaving it in mode=IR-REMOTE
- prox: after setting Infraredsensor to mode=IR-PROX (using a further python script)

Conclusion
- Setting the Infraredsensor to mode IR-PROX reduces the interrupts to sensible rate, without a relevant performance impact. 
- The infraredsensor in mode=IR-REMOTE is a slow device, I do not understand, that it uses nearly 6000 interrupts/second
- however, this problem occurs with Infraredsensor in **Input1**, in Input4 there is now slowup. The mindstorms user guide says, that input4 is the default for infraredsensor. I didn't find a restriction. 

Is this a bug in ev3dev, or a hardware problem in my brick?

artpoz, 2019-04-18:
I've got two versions of ev3dev, jessie from 02.2017 and micropython from 04.2019.
I  boot ev3dev and run the same program on both versions and messure the time
https://gist.github.com/artpoz/d66f72a6092b580b9903d088dd6d1262

 **System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```
boot                  53 s.
ps4explor3r.py  16 s.
shutdown          17 s.

```
Image file:         ev3-micropython-v1.0.0-sd-card-image
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165343C8E0
BOARD_INFO_TYPE=main
```
boot                1.42 s.
ps4explor3r.py   32 s.
shutdown           17 s.

So  EV3 brick starts twice slower and my program starts twice slower than on the latest version :(

dlech, 2019-04-18:
Newer software is more complex and runs slower on the same hardware.

snaildos, 2021-05-29:
This is getting annoying. I made my whole code with roberta, ran the .py file and it takes **10 mins** for it to start. WHAT...? Is there a way I can like.. somehow make it run faster?