ev3dev/ev3dev #1200: Problems with on_to_position()

ndward, 2018-12-07:
Previously I had the impression that if on_to_position() is used to instruct a motor to turn to a position where it happens to already be located then this will freeze the program. I think that's been fixed, but I still have had lots of problems with the on_to_position() function, which often causes my scripts to freeze for other reasons. This is true for both the medium motor and the large motor and seems to be worse when one movement is immediately followed by another and when the speed setting is low (e.g. 30).

Here below is my test script. The script should cause the position to be printed 7 times, so if that is not the case it means the program froze. The position values are often not what one would expect - I suppose the motor has not yet settled. The position values are as expected when the sleep() statement is enabled.

This script measures the position before the move rather than after to highlight the fact that the position is zero each time the script is started, something that should be noted in the documentation. The exception is when the motor was disconnected and reconnected before the script was launched - that caused the position to appear as 9284 and caused the program to freeze while the motor span forever. Perhaps not forever - perhaps it was turning through 9274 degrees. A workaround for that problem is to specifically set the position to zero as the program starts, such as by enabling the corresponding command in the sample script.

#!/usr/bin/env python3
from ev3dev2.motor import OUTPUT_B, MediumMotor, LargeMotor
from time import sleep
from sys import stderr

#motor = MediumMotor()
motor = LargeMotor(OUTPUT_B)

def goto(pos):
    print('motor.position = '+ str(motor.position), file = stderr)
    motor.on_to_position(speed=30, position=pos)
    #sleep(0.5)

#motor.position = 0
goto(10)
goto(20)
goto(-20)
goto(-30)
goto(-40)
goto(-50)
goto(-60)

Here are some results:
Medium motor run #1, speed = 30, sleep disabled
motor.position = -24
motor.position = 9

Medium motor run #2, speed = 30, sleep disabled
motor.position = 0
motor.position = 14

Medium motor run #3, speed = 30, sleep disabled
motor.position = 0
motor.position = 12

Medium motor run #4, speed = 50, sleep disabled
motor.position = 0
motor.position = 13
motor.position = 25
motor.position = -18
motor.position = -37
motor.position = -47
motor.position = -57

Large motor run #1, speed = 30, sleep disabled
motor.position = 0
motor.position = 14

Large motor run #2, speed = 30, sleep disabled
motor.position = 0
motor.position = 18
motor.position = 27
motor.position = -26
motor.position = -38
motor.position = -46
motor.position = -58

Large motor run #3, speed = 30, sleep disabled
motor.position = 0
motor.position = 12

Large motor run #4, speed = 30, sleep enabled
motor.position = 0
motor.position = 11
motor.position = 20
motor.position = -20
motor.position = -30
motor.position = -40
motor.position = -50

Large motor, after disconnecting and reconnecting, speed = 30, sleep enabled
(The motor ran forever)
motor.position = 9284

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-11-30
Kernel version:     4.14.80-ev3dev-2.3.1-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340377E
BOARD_INFO_TYPE=main
```

PS It often seems to be the case, as it is today Dec 7 2018, that the Stretch image proposed on ev3dev.org is not the latest one, as proposed on https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/  Is there a reason why we would not want people to use the latest version?

dlech, 2018-12-07:
cc: @ev3dev/python

>  Is there a reason why we would not want people to use the latest version?

There is always the risk we might have broken something without knowing it, so we make snapshot images that can be tested by the more serious users (like you) to catch any problems before we push them out to the more casual users. If there is a serious usability issue discovered in the image we offer on the main downloads page, then we will look at fast-tracking the latest snapshot image. Currently, there are no such issues that I am aware of.

WasabiFan, 2018-12-07:
@ndward what's the actual issue you're seeing? Is there a problem if you don't unplug the motor? And when exactly are you unplugging it?

WasabiFan, 2018-12-14:
> using this function often causes the program to freeze

I'm unsure of why that would be. I can test this once I am back near my EV3 mid next week.

> disconnecting and reconnecting the motor causes the position to not equal zero which is otherwise always the case when a script is launched.

I don't expect the motor position to be zeroed when a script starts. The position will generally be zero when a new motor is plugged in or when the whole system restarts, but as far as I know there's nothing in our library's `Motor` that would do such a thing. (We do, however, reset motors on initialization of any motor group class.) I should note two things:

- Unplugging/re-plugging a motor after initializing a `Motor` (or any other class) for that port is not supported. If you do `motor = LargeMotor(OUTPUT_A)` then unplug the motor connected to port A, the `motor` object is no longer expected to work, even if you plug it back in. I can't tell if that is what you are doing or not.
- You should never depend on `position` starting at a particular value. If you check a motor's position, you should expect that from then on the relative difference of the new position value will be correct... but the absolute `position` is largely meaningless unless you set `position` yourself to a known value.

ndward, 2018-12-15:
I've never unplugged a motor while a script was running. The phenomenon I'm talking about (non-zero position values at launch e.g. 92984, possibly random?) occurs when I disconnect and reconnect a motor before launching the script. Otherwise, without disconnecting, the script always seems to launch with position equal to zero.

WasabiFan, 2018-12-20:
I can reproduce the freezing issue. It will be _less bad_ in the upcoming "beta 3" release of the Python library, but not 100% resolved. I opened https://github.com/ev3dev/ev3dev-lang-python/issues/583 to track this.

As for the position zeroing, I can't reproduce what you are seeing. Additionally, you should  always be zeroing the position anyway if you need it to be zero.