ev3dev/ev3dev #1204: Add updated version of libopenmpt0 to ev3dev repositories

JakubVanek, 2018-12-10:
Hi,

### What are you trying to do?
We are trying to load OpenCV libraries on ev3dev stretch. However, a bug in libopenmpt0 (dependency of ffmpeg) crashes the entire program (JVM in our case). From what I understand there is a big stack allocation and this causes problems.

https://github.com/opencv/opencv/issues/10080
https://github.com/opencv/opencv/issues/10157

This issue seems to be tracked in: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=892272. Somebody found the commits upstream used to fix the problem. They weren't applied to stretch though.

I have created a package containing the latest 0.2 version of libopenmpt0 from their official website. I have removed all debian patches beginning with `up*` and then I simply replaced the orig archive. The resulting library does not have the described problem.

I would like to include this upgraded library in ev3dev repositories. I realized that I should get an approval for this (it adds an entirely new version of the library). Alternatively, I can report this issue to Debian upstream and request the two patches to be backported to stretch as well.

### What did you expect to happen?

N/A

### What actually happened?

N/A

### What hardware and software are you using (including version numbers)?
* Recent ev3dev-stretch snapshot.
* `libopenmpt0 (0.2.7386~beta20.3-3+deb9u3) `
* any jri-11-ev3 version
* https://github.com/ev3dev-lang-java/examples/blob/master/opencv/src/main/java/example/opencv/HelloWorldCV.java

Thanks,

Jakub

JakubVanek, 2018-12-10:
I am going to report the bug through `reportbug`. I will report if there is any progress.

JakubVanek, 2018-12-10:
Upgrade of libopenmpt to a new upstream version is not necessary - the patch described in the old bugreport works for us too.

jabrena, 2018-12-10:
Tomorrow, I will test the package and complex process.

JakubVanek, 2018-12-10:
I would flip the first two, so that the distro version gets overwritten by the fixed version:
1. Install OpenCV for Java `sudo apt install libopencv2.4-java`.
2. Install this package via `sudo dpkg -i <path>`.
3. Run `example.opencv.HelloWorldCV`.

JakubVanek, 2018-12-11:
Bugreport submitted: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=916222

dlech, 2018-12-12:
I don't expect this bug to be fixed in official Debian Stretch - at least not any time soon. I have forked the package source at https://github.com/ev3dev/libopenmpt.

You can add the patch to this and release it.

A few notes:
* I would like to keep the package version as `0.2.7386~beta20.3-3+deb9u3ev3dev1`. When we fork packages like this, we just add `ev3dev1` to the version. (Additional versions, if needed, will be `ev3dev2` and so on.)
* `UNRELEASED` will need to be changed to `stable` in `debian/changelog` before building and uploading the package.
* You can (should) use `./debian/release.sh` to publish the package.
* Please push the `ev3dev-stretch` branch and the tag created by the script above to ev3dev git repo when publishing the binary package.

JakubVanek, 2018-12-12:
Yaay - it doesn't crash anymore!

This means this issue is fixed for ev3dev for now.

Thanks,

Jakub

closed 2018-12-12 by JakubVanek