ev3dev/ev3dev #1205: two irq's consume 12% of CPU

magicalpear, 2018-12-11:
### What are you trying to do?
Booted up and ssh logged in, top showed two processes consumed 9.3% ( irq/249-ev3-tac) and  5.1% (irq/247-ti-ads7) of CPU time

### What did you expect to happen?
no significant CPU usage without loading programs

### What actually happened?

### What hardware and software are you using (including version numbers)?
latest ev3dev2 image 

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-11-13
Kernel version:     4.14.80-ev3dev-2.3.1-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534EC286
BOARD_INFO_TYPE=main
```

magicalpear, 2018-12-11:
when no activity (such as apt), the cpu consumption reduced to 6% and 4%

WasabiFan, 2018-12-11:
I'm not 100% sure, but my assumption has been that those were USB-related and a result of the Wi-Fi dongle sustaining its connection.

dlech, 2018-12-11:
These are kernel threads for the tachometers and the analog/digital converter.

It doesn't seem like the tachometer should be using quite that much CPU, but I would have to go back an look at the driver to see. It could be that the motor drivers are still polling the tachometer values every 2ms for the speed/position control.

The analog/digital converter is actually much improved over previous kernels. It use to use 10-15% all the time. The A/DC is used for identifying motors and sensors, monitoring the battery and, of course, analog sensor values.

magicalpear, 2018-12-13:
even for A/DC for the purpose of sensing and monitoring, it seems quite excessive. Maybe some polling interval to tune?

When the system is access files, it seems quite slow. Is the SD card very slow in that regard?

dlech, 2018-12-13:
> even for A/DC for the purpose of sensing and monitoring, it seems quite excessive. Maybe some polling interval to tune?

https://github.com/ev3dev/ev3dev/wiki/Internals:-ev3dev-stretch#changing-the-polling-rate

> When the system is access files, it seems quite slow. Is the SD card very slow in that regard?

https://github.com/ev3dev/ev3dev/wiki/Selecting-a-microSD-card

magicalpear, 2018-12-14:
Thanks for the links. For the A/DC, the default is 100Hz sampling rate. It seems 5% of 300MHz CPU is quite a lot for 100Hz sampling rate. Do you know if DMA/irq/or polling is used for SPI accessing ADC chip? Where I can find the related code for the ADC SPI driver? Thanks.

dlech, 2018-12-14:
It is using DMA (switching to DMA is how the usage was reduced from 10% to 5%).

* The ADC driver is [here](https://github.com/ev3dev/ev3-kernel/blob/ev3dev-stretch/drivers/iio/adc/ti-ads7950.c)
* The SPI driver is [here](https://github.com/ev3dev/ev3-kernel/blob/ev3dev-stretch/drivers/spi/spi-davinci.c)
* The tacho driver is [here](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/ev3/ev3_pru.c)