ev3dev/ev3dev #1206: Is it possible to shutdown an EV3 Motor/Sensor to save energy?

jabrena, 2018-12-11:
Hi,

I would like to know if EV3 Motor/Sensor are possible to be disabled by code to save energy.

Juan Antonio

JakubVanek, 2018-12-11:
I have done some searching in the EV3 HDK and the 5V power to sensors and motors cannot be disabled. However:
* It might be possible to send some shutdown command to the sensor. This however might require changes in the ev3 sensor link layer (tty line discipline?) if there is some sensor timeout.
* 5V on EV3 motors is used only for powering their tachometers.
* Power to the motor actuators themselves can be cut off completely by running the "coast" command.

Then there's the thing if it is worth it - one would need to measure this. Putting a current meter on the power line between EV3 and the sensor may be enough, if that handles possible power spikes well.

JakubVanek, 2018-12-11:
Hmmm, actually there is an issue about a similar topic already: https://github.com/ev3dev/ev3dev/issues/313

dlech, 2018-12-11:
> the 5V power to sensors and motors cannot be disabled.

Technically, it can be turned off (I've done it :wink:). However this 5V line is shared with the USB host port and the AD/C chip. I think it is actually *supposed* to turn off automatically if there is over 1A being used. I don't remember if I actually tried shorting it out to ground on the official LEGO firmware to see if it works there or not.

JakubVanek, 2018-12-11:
Oki, I have totally missed the circuitry on EN line and that there is a P_EN signal. Thanks for the correction.