ev3dev/ev3dev #1207: EVB img creation problems

michaelwareman, 2018-12-21:
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

michaelwareman, 2018-12-21:
Hello:
I followed the instructions of the EV3DEV.org website for creating a boot image for my EVB 1.0 cape with a BeagleBone Black. I am using a 32GB SanDsik Ultra. When I created the image Etcher creates three partitons, 2 I cannot see and get format message, the small partition has only the uEnv.txt file. I uncommented the 2 lines for the EVB cape. When I insert the card into the EVB and turn it on the screen lights up but nothing else happens.
I have a 8GB memory card with the EVB img on it and the unit boots up properly.
Any help would be greatly appreciated.

michaelwareman, 2018-12-21:
Update 
I finally found an older image — ev3dev-jesse-2017-09-14. This version works just fine. If you need more details please let me know. 
Mike

dlech, 2019-01-28:
Which image file did you use?

> When I created the image Etcher creates three partitons,

There should only be 2 partitions

>  I uncommented the 2 lines for the EVB cape.

There should only be one line that should be changed.

michaelwareman, 2019-01-28:
Hello David:
Both my EVB and the EV3 images boot up to the brickman. The problem was with the editor I was using. If I use wordpad then it does not work. If I use notepad (everything is one long line) and make the required edit it works.
Mike,

dlech, 2019-01-28:
Thanks for the update. Does that mean we can close this issue?

michaelwareman, 2019-01-28:
yes, but now have a new issue. see new comment
Mike,

closed 2019-01-28 by dlech