ev3dev/ev3dev #1208: Issues with the hitechnic color sensor

FalconFX9, 2018-12-24:
This is my code, I just want it to show the value for white.

from ev3dev2.auto import *
from time import sleep

mA = Motor(OUTPUT_A)
cl = Sensor('in2:i2c1')
cl.mode = 'WHITE'
while True:
    print(cl.value())

### What are you trying to do?
I'm trying to change the mode of the hitechnic color sensor v2

### What did you expect to happen?
I expected the mode to change

### What actually happened?

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 248, in _set_attribute
    attribute.write(value)
OSError: [Errno 22] Invalid argument

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "run_motor.py", line 6, in <module>
    cl.mode = 'WHITE'
  File "/usr/lib/python3/dist-packages/ev3dev2/sensor/__init__.py", line 179, in mode
    self._mode = self.set_attr_string(self._mode, 'mode', value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 290, in set_attr_string
    return self._set_attribute(attribute, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 251, in _set_attribute
    self._raise_friendly_access_error(ex, name)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 268, in _raise_friendly_access_error
    chain_exception(ValueError("One or more arguments were out of range or invalid"), driver_error)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 38, in chain_exception
    raise exception from cause
ValueError: One or more arguments were out of range or invalid

### What hardware and software are you using (including version numbers)?

Kernel : 4.14.80-ev3dev-2.3.1-ev3
Hardware : EV3, Hitechnic Color Sensor v2

FalconFX9, 2018-12-24:
Oh, my mistake, but I changed it to this :

cl.mode = 'RGB', 'value3'

But then it gives me this :

Traceback (most recent call last):
  File "run_motor.py", line 6, in <module>
    cl.mode = 'RGB', 'value3'
  File "/usr/lib/python3/dist-packages/ev3dev2/sensor/__init__.py", line 179, in mode
    self._mode = self.set_attr_string(self._mode, 'mode', value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 290, in set_attr_string
    return self._set_attribute(attribute, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 251, in _set_attribute
    self._raise_friendly_access_error(ex, name)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 256, in _raise_friendly_access_error
    raise driver_error
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 248, in _set_attribute
    attribute.write(value)
TypeError: a bytes-like object is required, not 'tuple'

WasabiFan, 2018-12-24:
Remove `, 'value3'`. `mode` is a string property.

FalconFX9, 2018-12-24:
Oh that fixed it, thanks a lot (I'm still a noob at this, sorry)

closed 2018-12-24 by FalconFX9