ev3dev/ev3dev #1209: no connections

rburtonresearch, 2018-12-26:
### What are you trying to do?
I have just flashed ev3 stretch beta and I am trying to establish connections. The 'all network connections' is blank, the status is 'no connection' the bluetooth says 'not available' and the wiFi is also 'not available'. 

### What did you expect to happen?
I expected to be able to connect to my computer via a network connection

### What actually happened?
no connections seem to be available.

### What hardware and software are you using (including version numbers)?
Lego Ev3 brickman v0.10.0
4.1480 ev3dev-2.3.1-ev3

rburtonresearch, 2018-12-27:
I would be happy just to use the VS code extension. 

However that is another issue. VS Code can 'see' the Ev3 via the 'PC' lead and it will play the "hello.py" program (from the VS code hello python master) but when I tried the 'demo.py' it came up with the error messages (below)
Code:
#!/usr/bin/env python3
from ev3dev.ev3 import *
import os
os.system('setfont Lat15-TerminusBold14')
mL = LargeMotor('outB'); mL.stop_action = 'hold'
mR = LargeMotor('outC'); mR.stop_action = 'hold'
print('Hello, my name is EV3!')
Sound.speak('Hello, my name is EV3!').wait()
mL.run_to_rel_pos(position_sp= 840, speed_sp = 250)
mR.run_to_rel_pos(position_sp=-840, speed_sp = 250)
mL.wait_while('running')
mR.wait_while('running')

Error messages
[pylint] Unable to import 'ev3dev.ev3' (2,1)
[pylint] Undefined variable 'LargeMotor' (5,6)
[pylint] Undefined variable 'LargeMotor' (6,6)
[pylint] Undefined variable 'Sound' (8,1)

dlech, 2018-12-27:
It sounds like you are running the demo.py program on your PC instead of the EV3.

WasabiFan, 2018-12-29:
In the bottom right corner of vscode, click "CRLF" and change it to "LF". Then try again.

rburtonresearch, 2018-12-29:
Thank you. 
Once I changed from "CRLF" to "LF", the "demo.py" program worked perfectly! It worked both from using 'F5' and running from the brick.

The other problem I have (relating to connections -see above-) has partially resolved itself as well.
I can now see Bluetooth options and have tried connecting to my computer. My computer reports a successful connection (ev3dev paired). However the brick reports errors when I try to connect.

GDBus.Error.org.bluez.Error.Failed.Protocol not available.

and I tried connecting from'all network connections and I got:
GDBus.Error.net.connectionError.Failed. input/output error