ev3dev/ev3dev #1210: ENOMEM when using I2C sensors with HiTechnic Sensor Multiplexer

dlech, 2018-12-28:
In https://github.com/ev3dev/ev3dev-lang-python/issues/586, @w1ref1re wrote:

> If I run `dmesg | tail` there are the lines
>
>     [ 1277.145774] ht-nxt-smux-i2c-sensor: probe of ev3-ports:in2:i2c8:mux1:lego-nxt-us failed with error -12
>     [ 1277.146681] lego-port port12: Added new device 'ev3-ports:in2:i2c8:mux1:lego-nxt-us'

dlech, 2018-12-28:
Totally forgot to allocate the memory for `data->sensor.mode_info` here: https://github.com/ev3dev/lego-linux-drivers/blob/3a92db95671708319f3c6a33cd044cb23e7e782a/sensors/ht_nxt_smux_i2c_sensor.c#L93-L97

Compare to: https://github.com/ev3dev/lego-linux-drivers/blob/3a92db95671708319f3c6a33cd044cb23e7e782a/sensors/nxt_i2c_sensor_core.c#L231-L236

w1ref1re, 2018-12-29:
Can I fix this on my own? If yes can I have some explanation how to do it please?

dlech, 2018-12-29:
You would need to build your own kernel to fix this. You can find info at https://github.com/ev3dev/ev3dev-buildscripts/#system-requirements

And hopefully it should be obvious (from the links above) which line you need to copy and paste to actually fix the problem.

w1ref1re, 2018-12-29:
Thank you. The line was obvious I just wondered where i can find the file. I will give it a try thanks.

w1ref1re, 2019-01-20:
I made it work and the 
```
SMUX
```
 detects the 
```
Ultrasonic sensor
```
 now. 
But it is not updating the values... It always displays 255. 

The code: 

```
us = Sensor(address="in2:i2c8:mux3" , driver_name = lego-nxt-us") 

while True:
    print(us.value(0))
```

dlech, 2019-01-28:
Fix released in `ev3dev-2.3.2-ev3` kernel.

closed 2019-01-28 by dlech