ev3dev/ev3dev #1216: EV3 sensor on HT mux not updating

w1ref1re, 2019-01-21:
I edited my kernel like described in #1210. The EV3 detects the sensor now and i can use it in my programm.
But however the sensor is not updating its values. I am always getting ```255``` as value no matter what sensor is connected to the mux. 

Here is my code:
```
us = Sensor(address="in2:i2c8:mux3" , driver_name = "lego-nxt-us") 
while True: 
    print(us.value(0))
```

Output :
```
255
255
255
255
...
```

dlech, 2019-01-21:
Could you send a pull request with that change? Then I'll try to have a look at it in the next week or so.

dlech, 2019-01-28:
Fixed in https://github.com/ev3dev/lego-linux-drivers/commit/930a506be5e69394f8cd2f6876c9a81b1efded45

dlech, 2019-01-28:
Fix released in `ev3dev-2.3.2-ev3` kernel.

closed 2019-01-28 by dlech