ev3dev/ev3dev #1217: brickpi3: sign of motor position gets flipped

dwalton76, 2019-01-25:
### What are you trying to do?
I am trying to set the position for a motor but the sign is getting flipped
```
robot@ev3dev:~/lego-crane-cuber$ echo "-21" > /sys/class/tacho-motor/motor0/position
robot@ev3dev:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor0/position
21
robot@ev3dev:~/lego-crane-cuber$
robot@ev3dev:~/lego-crane-cuber$ echo "21" > /sys/class/tacho-motor/motor0/position
robot@ev3dev:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor0/position
-21
robot@ev3dev:~/lego-crane-cuber$
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-12-14
Kernel version:     4.14.80-ev3dev-2.3.1-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dlech, 2019-01-25:
Probably something wrong with the logic here:

https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/brickpi3/brickpi3_ports_out.c#L216

dlech, 2019-01-28:
Fix released in `ev3dev-2.3.2-ev3` kernel.

closed 2019-01-28 by dlech