ev3dev/ev3dev #1218: brickpi3: overloaded state for motor not cleared by stop or reset

dwalton76, 2019-01-25:
### What actually happened?
Once the motor gets in `overloaded` state it seems to be stuck there...I cannot clear it by stopping the motor or reseting the motor.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-12-14
Kernel version:     4.14.80-ev3dev-2.3.1-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dwalton76, 2019-01-25:
On a related note, I often (probably always) see this msg whenever a motor is overloaded
```
Broadcast message from root@ev3dev (somewhere) (Fri Jan 25 13:55:17 2019):

Low battery. Power off or connect a charger soon.
```

I am plugged into the wall though so power/battery is fine.

dlech, 2019-01-25:
> On a related note,

Probably not. New issue please. And state what the actual voltage reading is.

dlech, 2019-01-25:
> ..I cannot clear it by stopping the motor or reseting the motor.

This may be a BrickPi3 firmware bug. This could be confirmed by using the official DexterOS and see if you get the same results.

dlech, 2019-01-27:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/brickpi3.html#battery

dlech, 2019-01-27:
So probably: 

    cat /sys/class/power_supply/brickpi3-battery/voltage_now

dlech, 2019-01-28:
Please update to the latest BrickPi3 firmware version (currently 1.4.6)

    sudo update-brickpi3-fw

dwalton76, 2019-03-24:
yep this is fixed now, am running

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-12-14
Kernel version:     4.14.94-ev3dev-2.3.2-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

closed 2019-03-24 by dwalton76