ev3dev/ev3dev #1219: Is there a definitive fix to gyro drift?

thFury, 2019-01-26:
Hello. I was wondering if this issue has been fixed. From what I understand, the gyro starts drifting when the sensor is moved during boot (this has been confirmed), though sometimes this happens randomly.

The 'best' way would be to reboot the sensor on every program launch, though I am not sure if this is possible without hardware modifications, + eventual backup plans if the gyro is still not working.

Other ways that I found here are switching port modes: (not tested yet, brick not in hand)

`set_port_mode(sn_gyro, INPUT_OTHER_UART);
  set_port_mode(sn_gyro, INPUT_EV3_UART);`

And switching sensor modes (this resets the gyro, though if it's drifting it will keep on drifting..).

On StackOverflow I also found someone saying to sample d/s x times per second and calculate based on that. I am not sure if this will 'bypass' the drift completely or if the drift is still present (as said, brick not in hand).

dlech, 2019-01-28:
IMHO, changing the port mode to force the gyro sensor to reset is the best solution. You just need to be sure that the sensor is **completely** still when you do this.

Switching modes only works for older gyro sensors (date stamp of xxN2 or xxN3).

thFury, 2019-02-09:
Alright. I've found the root cause of the drift.

Basically, when the brick boots up, it does the initial sensor calibration via GYRO-CAL.
If the sensor was moving during this calibration, the brick calculates a wrong d/s, creating the drift.

Fortunately, this drift seems deterministic (I've found it varies by +- 1 sometimes), and knowing this you could now build a function that takes the drift into consideration.

An even better solution is to call GYRO-CAL again during program execution, making sure that the sensor is still. 

Currently, I've tested and retested this method 5+ and it seems to be reliably fixing the drift.

It would be nice to add this to the docs (GYRO-CAL says "calibration ???" on there), to help other users out.

haltonstemclub, 2021-01-19:
Hi @thFury , is it possible to execute the GYRO-CAL in Micropython? If so, what would the code by?

WaveSTORM2023, 2023-10-19:
> Hi @thFury , is it possible to execute the GYRO-CAL in Micropython? If so, what would the code by?

Wondering the same thing. I fixed the issue in scratch and labview, by tricking the gyro into thinking it was an ultrasonic sensor, but Micropython is more smart, and it reads as an error because an ultrasonic sensor isn't in that port. I think I will just have to keep trying different ways.