ev3dev/ev3dev #1220: unable to import ev3dev2.

michaelwareman, 2019-01-28:
### What are you trying to do?
I am following the instructions from this website:
https://sites.google.com/site/ev3devpython/setting-up-vs-code

### What did you expect to happen?
My problem is that when I use sample scripts I get this error message:
Unable to import 'ev3dev2.motor'
I have done everything that this site tells me to do.
The VS code shows that there are LF at the end of each of my code lines. 
The F5 function does not work but I am able to download the code to to the EV3. I can run the file from the EV3

### What actually happened?

### What hardware and software are you using (including version numbers)?
I am using the current stretch version and ev3dev-browser 0.8.1, and Python 3.6.3 64 bit

dlech, 2019-01-28:
For VS Code, I like to set up a virtualenv. In the terminal in VS Code (should be same directory as your project) run the following.

Non-windows:

    python3 -m venv .venv
    source .venv/bin/activate
    pip install --upgrade pip
    pip install python-ev3dev2

Windows (make sure you are using CMD, not PowerShell):

    py -3 -m venv .venv
    venv\Scripts\activate
    pip install --upgrade pip
    pip install python-ev3dev2

Then there is a button down in the status bar of the editor that you can click to select your virtualenv for the editor as well. It looks like this:

[image]

And when you click it, select the `.venv` folder (the one you just created).

[image]

VS Code will probably ask you to install more packages in your virtualenv sooner or later for a Python linter and whatnot.

dlech, 2019-01-28:
FYI, the `.` at the beginning of the `.venv` folder name is important. It tells the ev3dev VS Code extension not to download the entire virtualenv to your EV3 (which would take a long time).

dlech, 2019-01-28:
Also see https://github.com/ev3dev/vscode-hello-python which has the proper settings in the `.vscode` directory to make <kbd>F5</kbd> work.

dlech, 2019-01-28:
It looks like I missed it too :blush:

michaelwareman, 2019-01-28:
I copied the lines as you typed them.
Now what? How do I fix this?

dlech, 2019-01-28:
I typed it wrong (missed the `.` in `.venv\Scripts\activate`), have a look again (I have fixed it).

michaelwareman, 2019-01-28:
can I shutdown the vs code program and start again in the morning?

dlech, 2019-01-28:
I don't see why not :wink:

michaelwareman, 2019-01-28:
Excellent! I will let you know how it goes tomorrow sometime.
Thank you for all your help!
Mike

michaelwareman, 2019-01-28:
Hello David:
I did the commands as listed above. Just for your knowledge the command for upgrading pip is:
python -m pip install --upgrade pip 
which I d did after I was done. I now have the Python 3.7.1 32 bit ('.venv':venv) set in VS code. I still get the unable to import ev3dev.motor error message. Here is the line in the code I am using:
from ev3dev2.motor import LargeMotor
My kernal version on the EV3 is: 4.14.80-ev3dev-2.3.1-ev3

dlech, 2019-01-29:
> python -m pip install --upgrade pip

Thanks!

> import ev3dev.motor

It looks like you are missing the `2`.

> from ev3dev2.motor import LargeMotor

works for me :man_shrugging:

michaelwareman, 2019-01-29:
Sorry, I miss typed the from lines do have the 2 in it:
{
	"resource": "/c:/Users/Wareman/Documents/vscode-hello-python-master/motors.py",
	"owner": "python",
	"code": "E0401",
	"severity": 8,
	"message": "Unable to import 'ev3dev2.motor'",
	"source": "pylint",
	"startLineNumber": 3,
	"startColumn": 1,
	"endLineNumber": 3,
	"endColumn": 1
}
Interestingly, I was trying to see if I understood the use of the While True: statement that if I Comment the line than the error for import disappears and an indent error appears. If I fix the indent than the import error reappears.  in case this matters this is the import statement line I am using:
from ev3dev2.motor import LargeMotor
from ev3dev2.motor import SpeedDPS, SpeedRPM, SpeedRPS, SpeedDPM
The does work on the EV3 -- It appears that the EV3 creates an executable. 
Mike,

Yash10bloodstorm, 2019-10-21:
I copy pasted the same thing it worked the last time but not this time, the .venv foldefr doesnt show up when i click that button on the status bar.
Also is there a way to permanently set the virtual environment.

SzoDavid, 2019-10-22:
When I ran
py -3 -m venv .venv
It returned
Error: Command '['F:\\EV3\\WRO2019\\.venv\\Scripts\\python.exe', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 3221225595.

WasabiFan, 2019-10-25:
You should be able to use Python directly rather than through this "py" wrapper.

SzoDavid, 2019-10-25:
I fixed the problem by reinstalling Python
But I can't select the virtualenv, there is no button in the status bar, which would open that window

Yash10bloodstorm, 2019-11-05:
@dlech  It was working pretty smoothly, but now even though I've selected the .venv and all that seems to be correct, but still the syntaxes for lego are still red-underlined 
[image]

NotUnusual, 2020-01-10:
I've had the same Problem.

Fore me it worked to put the 'vscode-hello-python-master' file in another folder. In the beginning this folder was in the C:\Users\fbk\Documents folder. But the system had permission issues. So I put it under D:\programs\ev3dev2. In the following step I set up a virtual environment in a slight different way than dlech recommended. I typed these 4 lines in the vs code Terminal:

py -3 -m venv .venv
.venv\Scripts\activate
python -m pip install --upgrade pip
pip install python-ev3dev2

As this worked for my windows system, this is the code for non windows systems:

python3 -m venv .venv
. .venv/bin/activate
pip install --upgrade pip
pip install python-ev3dev2

It's also slightly different to dlechs version.

Hope this works for you

appleoddity, 2020-11-13:
I've watched the videos and read through the instructions found here:
https://sites.google.com/site/ev3devpython/setting-up-vs-code
and here:
https://github.com/ev3dev/vscode-hello-python

This includes creating the virtual environment. This is on Windows with the latest version of VS Code and Python 3 64-bit installed today.

I am assuming that the instructions at ev3python are outdated and instead of using `from ev3dev.ev3 import *` I used `from ev3dev2 import *` which resolved the missing module error.

Yet, I still get the error that 'LargeMotor' is not define when I try to use `mL = LargeMotor('outB')`, if I run the code.

Can someone please update this thread with instructions that are current and working?

Orestis-Papadopoulos, 2020-11-27:
Hello everyone.
I recently faced the problem with 'ev3dev2' ModuleNotFoundError and I solved it (in Windows) by copying the directory named 'ev3dev2' from C:\Users\User\AppData\Local\Programs\Python\Python38-32\Lib to the current working directory of Python. 

1. Go to the Python interpreter and find the current working directory:
>>>import os
>>>os.getcwd()
'C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Python37_64'

2. Copy the 'ev3dev2' directory the path above.

3. After that, the module should be visible in the module list. Confirm this by typing in the interpreter:
>>>print(help('modules'))
The output is a long list including the modules you can successfully import, I think.

*Another mistake I made was that I added comments above the shebang. For some reason this is wrong. Always make sure the shebang (#!/usr/bin/env python3) is at the first line of your code.

I hope this helps.