ev3dev/ev3dev #1224: Finding the MAC address

CallumJWilson, 2019-02-14:
Similar to d-e-t's issue [here](https://github.com/ev3dev/ev3dev/issues/908), I am trying to connect my BrickPi to a university wifi network. To do this I must get the device's wifi MAC address whitelisted on the network, however I cannot work out how to find this MAC address.

In the previously mentioned thread people said to find the MAC address on the usb wifi dongle's label, but mine does not have this information. I am currently connecting to the raspberry pi via an ethernet connection with a SSH terminal.

The hardware is a BrickPi v2.8 on a Raspberry Pi 2, with a Vilros 802.11n usb wireless adapter. I have the latest ev3dev-stretch image installed for the device.

dlech, 2019-02-14:
You could try running `ifconfig` in a terminal.

CallumJWilson, 2019-02-15:
Yep that works thanks! Just needed to make sure I enter `sudo ifconfig` and it is labelled as 'ether' for both the ethernet and wireless connections

closed 2019-02-15 by CallumJWilson