ev3dev/ev3dev #1225: Network-Manager Installation

ashami81, 2019-02-18:
### What are you trying to do?
I'm trying to install network-manager package on Debian stretch to have 3G support on Lego Mindstoms

### What did you expect to happen?

### What actually happened?
after running the command **sudo apt-get update && sudo apt-get install network-manager curl**

The command gets stuck and I lose ssh connectivity to ev3 always at this point, 

Processing triggers for libc-bin (2.24-11+deb9u3) ...
Processing triggers for systemd (232-25+deb9u6) ...

### What hardware and software are you using (including version numbers)?

HW: LEGO MINDSTORMS EV3 with firmware: 1.10E

OS: ev3dev-stretch-ev3-generic-2018-11-13
[apt-get_output.txt](https://github.com/ev3dev/ev3dev/files/2874578/apt-get_output.txt)

dlech, 2019-02-18:
ev3dev uses ConnMan for network management, so installing NetworkManger uninstalls ConnMan and breaks the network connection.

You should be able to install the `ofono` package and use a 3G network with ConnMan without installing NetworkManager.

ashami81, 2019-02-19:
Thanks a lot, I'll try it out