ev3dev/ev3dev #1226: shutdown EV3

MarkSLF, 2019-02-21:
### What are you trying to do?
I can connect the EV3 with the Open Roberta Lab via wifi. The transfer and running of programs works fine. But after I transfer a program to the EV3 brick the LCD of the EV3 becomes empty.  After this I have no chance to go to the menu of the EV3 to shutdown / power off the brick. So I have to take out the batteries every time when I want to finish the work with the EV3.

How can I solve this ? What can be the problem?
     

### What hardware and software are you using (including version numbers)?
Lego EV3
ev3dev-stretch beta2 and ev3dev-jessie
Open Roberta Lab

BertLindeman, 2019-02-21:
It is not a remedy to your problem, but instead of removing the battery, you might reboot with the keystrokes documented in this [video by LEGO](https://www.lego.com/en-us/mindstorms/videos/how-to-reset-the-ev3-p-brick-fbcbdbed398e4e12a7ce30fa662c54be)
For some EV3 models easier than dismantling your robot.

dlech, 2019-02-21:
cc: @ensonic

ensonic, 2019-02-23:
@MarkSLF You can press center + down to terminate a program. Could you attach a screenshot of the programm (or export the xml in the open-roberta lab) so that I can check what program is causing this?

Also please keep using ev3dev-jessie with openroberta for now.