ev3dev/ev3dev #1227: gdbserver removed from snapshot stretch images

ThadHouse, 2019-02-22:
At least as of the 2019-02-18 snapshot for ev3, gdbserver is not included anymore in the default image. This is something that should always be included with any image, as even if the device cannot run gdb itself, it should be able to have remote debugging connect to it at any time. Not having gdbserver means if an issue occurs, the device must be connected to the internet in order to download the debugger, which is not a fast process. Please add this back in, as it is a very useful package.

```
Image file:         ev3dev-stretch-ev3-generic-2019-02-18
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534626AC
BOARD_INFO_TYPE=main
```

dlech, 2019-03-04:
Fixed in 2019-03-03 snapshot image.

closed 2019-03-04 by dlech

ThadHouse, 2019-03-04:
@dlech as a quick question, when would be an ETA on a final stretch image? We are interested in using the for the WRO competition, and would be nice to have a release image if possible, rather then snapshots.

dlech, 2019-03-04:
This snapshot (2019-03-03) will probably become the next release, so please test it.

ThadHouse, 2019-03-04:
Ok, I absolutely will. We build our own cross c++ toolchain and deployment setup, so it's a good test of does everything work.