ev3dev/ev3dev #1231: EV3 not recognizing mindsensors.com Light Sensor Array

KevinRSiu, 2019-03-18:
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

dlech, 2019-03-19:
My light array sensor is working, so I'm guessing that we have a different firmware version in the sensor. If you put the port in `other-i2c` mode and then run `sudo i2cdump -y -r 0x00-0x17 3 0x0A` where `6` is the number of the input port plus two, hopefully it will do something.

KevinRSiu, 2019-03-19:
Okay, I used that command and this is the result I get:
```
robot@ev3dev:~$ sudo i2cdump -y -r 0x00-0x17 5 0x0A
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX                            XXXXXXXX
```
For reference, this is the command's result when the sensor is not connected. 
```
robot@ev3dev:~$ sudo i2cdump -y -r 0x00-0x17 5 0x0A                             
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
10: 00 00 00 00 00 00 00 00                            ........
```
So, it seems to be reading or detecting _something_, although I'm not sure what it is. I tried restarting the EV3 and connecting the sensor in different ports, and the result is the same every time.

dlech, 2019-03-19:
Have you changed the I2C address of the sensor from the default address? You can scan all addresses like this (where 5 would be input port 3):

    sudo i2cscan 5

KevinRSiu, 2019-03-19:
I haven't changed its address. This is the result of the command. (I checked that the result is the same in different ports, even restarting the EV3).  
```
robot@ev3dev:~$ sudo i2cdetect 5
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5.
I will probe address range 0x01-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:    -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
For reference, this is what the reading is when there is nothing connected to the port. 
```
robot@ev3dev:~$ sudo i2cdetect 5
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5.
I will probe address range 0x01-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:    01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```

dlech, 2019-03-19:
Do the lights on the sensor turn on when you plug the sensor in or when you probe it?

I'm out of good ideas (you've checked all of the obvious stuff like possible broken wires), but you could check the output of `dmesg` for anything suspicious. You could try upgrading the kernel (`sudo apt update && sudo apt upgrade` - EV3 must be connected to the internet and it takes a long time).

KevinRSiu, 2019-03-19:
The lights on the sensor turn on for a while when it's connected, but they go off shortly after. This has happened too when connected to the EV3 without ev3dev so it seems there's no problem there. The lights don't turn on when probing the sensor with the ` i2cdetect` command.

I checked `dmesg`, but it doesn't produce any suspicious messaging. Just commands like the similar:
```
[ 5704.166528] lego-port port2: Added new device 'ev3-ports:in3:nxt-i2c-host'
[ 5760.521383] lego-port port2: Removed device 'ev3-ports:in3:nxt-i2c-host'
```
I have tried to connect the sensor while being root, and the command `echo 'ms-light-array 0x0a' > /sys/bus/i2c/devices/i2c-5/new_device` has worked, but the sensor doesn't light up and all the readings are `0.` When trying to put a command in Brickman, the following error appears. 

[image]

And, when probing the sensor, the following information appears. 

```
root@ev3dev:/home/robot# sudo i2cdetect 5
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5.
I will probe address range 0x01-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:    -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
Also, I did upgrade the kernel but nothing happened, sadly. It could be a problem with the sensor, but I find it odd that it does work with the EV3-G blocks and not here.

dlech, 2019-03-19:
I'm afraid I'm out of ideas unless you happen to have a logic analyzer.