ev3dev/ev3dev #1233: My ev3 brick will not start

lizisawizard, 2019-03-22:
### What are you trying to do?
I'm using the ev3dev and program my robot in python. 

### What did you expect to happen?
Before, it started fine. It displayed the proper screen and it connected to visual studios ok.

### What actually happened?
Instead, it started, but didn't flash the expected amber light and displayed “Alert! /dev/mmcblk0p2 does not exist. Dropping to a shell!”. 

### What hardware and software are you using (including version numbers)?
I'm using the ev3dev, Python and Visual Studios.

I also had a problem getting my robot to move when it did start. 
It kept saying "LargeMotor(outA) is not connected". 
If someone could help with both problems it'd be greatly appreciated.

dlech, 2019-03-22:
Do you have another microSD card you can try?

lizisawizard, 2019-03-23:
@dlech Hello! I got a new microSD and it works again! Thank you! I am still getting the "LargeMotor(outB) is not connected" traceback. Do you know how to fix that?

dlech, 2019-03-24:
Can you post your program? Have you eliminated the possibility of a bad wire or a bad motor or simply a wire not plugged in all the way?

lizisawizard, 2019-03-25:
`#!/usr/bin/env python3
from ev3dev2.motor import LargeMotor,OUTPUT_B,OUTPUT_C   #we're importing the motor function
from ev3dev2.motor import SpeedDPS, SpeedRPM, SpeedRPS, SpeedDPM #we're importing other functions related to speed
from time import sleep

a = LargeMotor(OUTPUT_B)
d = LargeMotor(OUTPUT_C)

dists = [20,60,80,50]

for r in dists:
    j = ((.0241*distances(r)) - .3596);
    if (j % 2 == 0):
        a.on_for_seconds(speed=SpeedRPS(1), seconds=(-j))
        d.on_for_seconds(speed=SpeedRPS(1), seconds=(-j))
        sleep(2)
    else:
        a.on_for_seconds(speed=SpeedRPS(1), seconds=(j))
        d.on_for_seconds(speed=SpeedRPS(1), seconds=(j))
        sleep(2)`

I deducted the motors and wires are working properly and they are plugged in all the way.

dlech, 2019-03-25:
It works for me (although I get an error later about `distances` not being defined).

Are the motors listed in the _Device Browser_ on the EV3?

lizisawizard, 2019-03-25:
They are, could it just be my brick that is the problem? I have an alternate

dlech, 2019-03-25:
What address to the motors have in the _Device Browser_? (it should say something like `lego-ev3-l-motor at ev3-ports:outB`)

lizisawizard, 2019-03-25:
Yeah the device browser shows that. I plugged in the new brick and it is outputting the same error. Every time I run the code a file called 'Subtask2.py.err.log' appears.

lizisawizard, 2019-03-25:
IT WORKS NOW!
i had to change clf to lf!

closed 2019-03-25 by lizisawizard