ev3dev/ev3dev #1235: Mixing  NXTServo-v3 with a multiplexer ?

cho934, 2019-03-25:
Hello,

### What are you trying to do?
Is it possible to connect the NXTServo-v3 to the EV3SensorMux ? or to the NSX2020 ?
What is the protocol of the NXTServo-v3 ? NXT/i2c or EV3/UART ?

### What did you expect to happen?
I have no more place on the EV3 brick for sensors, so I'm trying to use severals mux.

### What actually happened?
If NXTServo-v3 can be connected to a mux, how can I use it ? with which device name ?

dlech, 2019-03-25:
I haven't actually done this myself, but theoretically, you can change the address of the NXTServo-v3, so you can give each one a unique address. Then you can use the [Port Splitter for NXT Digital Sensors](http://www.mindsensors.com/ev3-and-nxt/52-port-splitter-for-nxt-digital-sensors) to attach more than one to a single port. Don't forget to [translate the I2C addresses in the user manual to/from Linux I2C addresses](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/i2c.html#useful-info). If you use the same addresses as the EV3 sensor multiplexer, then the servo controllers should be automatically detected when they are plugged in (assuming you make the connection to the EV3 _after_ connecting the servo controllers to the MUX). You can find the `address` of each device in the sensor browser on the EV3.

cho934, 2019-03-25:
ok.
if I understand well, the NXTServo-v3 communicate with NXT/i2c protocol.
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#ms-8ch-servo

So I can't use the EV3SensorMux but the  Port Splitter for NXT Digital Sensors.

So it didn't solve my problem. 
I have 4 LEGO EV3 Infrared to connect also and 2 NXT Angle Sensors. So I have 2 solutions:
1. I connect 2 EV3SensorMux end to end using only one EV3 port. is it possible with 4 "LEGO EV3 Infrared"?
2. I use the "Port Splitter for NXT Digital Sensors" to mix the 2 "NXT Angle Sensors" and the NXTServo-v3. In that case, is there any issues about time latencies with both "NXT Angle Sensors" ?

dlech, 2019-03-25:
If this is the HiTechnic Angle Sensor, then you can only use one on the multiplexer because as far as I know, you can't change the I2C address. Each device on the I2C mux needs to have a unique address.

However, the EV3 sensor mux is also an I2C device, so you could connect the EV3 sensor mux, one angle sensor and the servo controller to I2C mux (Port Splitter for NXT Digital Sensors).

I don't think you would notice any difference in latency on the I2C devices.

cho934, 2019-03-25:
And 2 EV3SensorMux devices connected "end to end" using only one EV3 port. Is it possible with 4 "LEGO EV3 Infrared" for instance ?

dlech, 2019-03-25:
No, the EV3SensorMux is an I2C device itself but only supports UART devices plugged into it. So you can't plug one EV3SensorMux into another one. You could connect two EV3SensorMux to one port using Port Splitter for NXT Digital Sensors if you change the I2C addresses of one of the EV3SensorMux.

cho934, 2019-03-26:
okidoki, I got it. Thanks!