ev3dev/ev3dev #1237: brickpi3 motor does not move for run-to-abs-pos

dwalton76, 2019-03-25:
### What are you trying to do?

Trying to use run-to-abs-pos on a brickpi3 to move a large motor.  The motor is at 270, I want it to move to 261, I set position_sp to 261 and do "run-to-abs-pos" but the motor never moves.

```
robot@brickpi3:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor3/position
270
robot@brickpi3:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor3/position_sp
261
robot@brickpi3:~/lego-crane-cuber$
robot@brickpi3:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor3/state

robot@brickpi3:~/lego-crane-cuber$ echo "run-to-abs-pos" > /sys/class/tacho-motor/motor3/command
robot@brickpi3:~/lego-crane-cuber$
robot@brickpi3:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor3/state

robot@brickpi3:~/lego-crane-cuber$ cat /sys/class/tacho-motor/motor3/position
270
robot@brickpi3:~/lego-crane-cuber$
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2019-03-03
Kernel version:     4.14.94-ev3dev-2.3.2-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2

BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dlech, 2019-03-25:
Works for me.

Is the `speed_sp` non-zero in your case?

dwalton76, 2019-03-25:
This is really weird, I unplug/plug the motor and run this multiple times via cut-n-paste and all is fine
```
echo 270 > /sys/class/tacho-motor/motor0/position
echo 261 > /sys/class/tacho-motor/motor0/position_sp
echo 114 > /sys/class/tacho-motor/motor0/speed_sp
echo "run-to-abs-pos" > /sys/class/tacho-motor/motor0/command
sleep 1
cat /sys/class/tacho-motor/motor0/position
```

but when I run the equivalent via python
```
robot@brickpi3:~/ev3dev-lang-python$ cat foo.py
#!/usr/bin/env python3

from ev3dev2.motor import OUTPUT_A, Motor
import logging

logging.basicConfig(level=logging.INFO,
                    format="%(asctime)s %(levelname)5s: %(message)s")
log = logging.getLogger(__name__)

motor = Motor(OUTPUT_A)
#motor.reset()
motor.position = 270
motor.position_sp = 261
motor.speed_sp = 114
motor.run_to_abs_pos()
robot@brickpi3:~/ev3dev-lang-python$
```

Most of the time it fails to move the motor
```
robot@brickpi3:~/ev3dev-lang-python$ ./foo.py
2019-03-25 22:05:54,640  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/position' mode='rb+' closefd=True>, name position, value 270
2019-03-25 22:05:54,642  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/position_sp' mode='rb+' closefd=True>, name position_sp, value 261
2019-03-25 22:05:54,643  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/speed_sp' mode='rb+' closefd=True>, name speed_sp, value 114
2019-03-25 22:05:54,643  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/command' mode='wb' closefd=True>, name command, value run-to-abs-pos
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
270
robot@brickpi3:~/ev3dev-lang-python$
```

Once it is in this state I cannot get it to move even when I take python out of the picture
```
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
270
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position_sp
261
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/speed_sp
114
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ echo "run-to-abs-pos" > /sys/class/tacho-motor/motor0/command
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
270
robot@brickpi3:~/ev3dev-lang-python$

```

dwalton76, 2019-03-25:
The log statements above should be
2019-03-25 22:05:54,640  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/position' mode='rb+' closefd=True>, name position, value 270
2019-03-25 22:05:54,642  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/position_sp' mode='rb+' closefd=True>, name position_sp, value 261
2019-03-25 22:05:54,643  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/speed_sp' mode='rb+' closefd=True>, name speed_sp, value 114
2019-03-25 22:05:54,643  INFO: Motor(spi0.1:MA): _set_attribute <_io.FileIO name='/sys/class/tacho-motor/motor0/command' mode='wb' closefd=True>, name command, value run-to-abs-pos

Maybe something about the `rb+` is corrupting a file?

dlech, 2019-03-25:
The difference is that the python code sets the `position` attribute. I was able to reproduce the problem after doing this.

For some reason, it looks like I decided that 10 degrees was close enough to the target when I wrote the brickpi3 driver.

https://github.com/ev3dev/lego-linux-drivers/blob/c1e25cc042a381f7d9b2c61274d7ec8b93a80ea9/brickpi3/brickpi3_ports_out.c#L474-L475

dwalton76, 2019-03-25:
The bash code was setting the position attribute too though...not seeing the delta.

dlech, 2019-03-26:
`cat` in the first comment doesn't set the value and I missed the bash in the later comment :smile:

dlech, 2019-03-26:
Also, you won't hit this bug if stop_action == hold.

dwalton76, 2019-03-26:
I still don't follow why I see different behavior when setting the attributes with python vs bash or why once it is in this broken state running the commands via bash doesn't move the motor either...but if you have a fix I am happy :)

dlech, 2019-03-26:
I was able to reproduce the problem using bash. Basically, if the difference between the current position and the setpoint was < 10, then the motor would not move except in rare cases when the kernel is really busy and the motor actually started moving before the position check was run.

dlech, 2019-04-08:
Fix released in kernel `ev3dev-2.3.3`.

closed 2019-04-08 by dlech

dlech, 2019-04-13:
This fix to this issue caused #1248. So reverted that fix and trying a new one.

dlech, 2019-05-09:
Fix released in ev3dev-2.3.4 kernel