ev3dev/ev3dev #1238: brickpi3 unplugging/plugging in motor decrements position attribute by 3 or 4

dwalton76, 2019-03-26:
### What are you trying to do?

This was with a large motor, everytime I unplug it and plug it back in the position decrements by 3 or 4 (most of the time it is 3).  There is an unplug/plug of the motor between each `cat` below

```
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
263
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
260
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
257
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
254
robot@brickpi3:~/ev3dev-lang-python$
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
251
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
247
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
244
robot@brickpi3:~/ev3dev-lang-python$
```
I do not see this on EV3 but it isn't exactly apples to apples because on EV3 when we unplug/plug the motor moves from ``motor0`` to ``motor1`` and then from ``motor1`` to ``motor2``.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2019-03-03
Kernel version:     4.14.94-ev3dev-2.3.2-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dlech, 2019-03-26:
What is the expected behavior? This sounds like reasonable behavior to me. When you plug/unplug the motor, I'm sure it creates electrical noise on the pins which could cause the tacho to count in either direction. BrickPi3 doesn't do automatic port detection like EV3. I would recommend doing the reset command any time you plug in a motor.

dwalton76, 2019-03-26:
If it is from the electrical noise on the pins wouldn't we see the same issue on EV3?

dlech, 2019-03-26:
Nope. On EV3, it can detect when a motor is plugged in. It waits 100msec after things have settled down before even starting to load the kernel driver.

dlech, 2019-03-26:
Although I suppose that you could watch the raw tacho motor value from the IIO driver and see what happens.

dlech, 2019-03-26:
Yup, I checked it and the EV3 varies by 1 count for me. My theory is that it has more resistors and stuff that dampen the noise compared to BrickPi.

dwalton76, 2019-03-26:
Gotcha

closed 2019-04-18 by dwalton76