ev3dev/ev3dev #1240: Bluetooth problems

YoitzWolf, 2019-04-02:
### What are you trying to do?
I would like to connect the ev3dev module with the typical ev3
And...nothing happened ... the connection failed (connection is OK about 1-2 seconds). What should I do?
### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

dlech, 2019-04-02:
Can you share your code?

WasabiFan, 2019-04-02:
@dlech I'm not sure there is any code for a Bluetooth issue... We do need to know the steps you took to see the error and what exactly the error was though.

usmanov-danil, 2019-04-14:
I have the same question. As I know in ev3dev doesn't exist a native way to do communication between two ev3 bricks. I didn't find any information about how to do it and what I should write in my code to solve it.