ev3dev/ev3dev #1241: mindsensors.com 8-channel Servo Controller - issue with set_rate_sp

cho934, 2019-04-04:
### What are you trying to do?
I tried to change each speed rate of each servomotors connected to the "ms-8ch-servo"

### What did you expect to happen?
Change the speed of each servo

### What actually happened?
Only the speed of the servo n°0 works, not the others. And modifying the rate speed of the servo n°1 change the speed rate of the servo n°2 !
I really don't understand the results...

### What hardware and software are you using (including version numbers)?

Ev3dev has just been updated.
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

Would it be possible to fix it as soon as possible ? We are using again ev3dev during the next robotic 2019 contests (in 2 weeks in Belgium, then in Switzerland, then in France )
Thanks in advance.

dlech, 2019-04-05:
ev3dev-jessie is no longer being updated. Please use ev3dev-stretch.

You could possibly work around the issue by putting the input port into `other-i2c` mode and use I2C commands directly.

cho934, 2019-04-05:
Thanks for the answer.

> ev3dev-jessie is no longer being updated. Please use ev3dev-stretch.

Does ev3dev-lang-cpp is ready for ev3dev-stretch ? or are there too many changes/bugs ?
I'm going to prepare a sdcard and try to test it this weekend.
Do you think the bug will still there using stretch ?

> You could possibly work around the issue by putting the input port into `other-i2c` mode and use I2C commands directly.
ok. 
Doc is there.
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#ms-8ch-servo

But where can I find what are the I2c commands for this device?

dlech, 2019-04-05:
> Does ev3dev-lang-cpp is ready for ev3dev-stretch ?

cc @ddemidov

> Do you think the bug will still there using stretch ?

Possibly. I don't think very many people use the servo controller.

> But where can I find what are the I2c commands for this device?

In the User's Guide from mindsensors.com.

[This page](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html) might also be helpful.

cho934, 2019-04-06:
> > Does ev3dev-lang-cpp is ready for ev3dev-stretch ?
> 
> cc @ddemidov

I'm currently testing it.

> 
> > Do you think the bug will still there using stretch ?
> 
> Possibly. I don't think very many people use the servo controller.

LoL. This device is very cool to add plenty of small servomotors ;)
The same issue is still there with set_rate_sp()  using ev3dev stretch.
Is it possible to fix it ? I'm ready to test it this weekend.

```
Image file:         ev3dev-stretch-ev3-generic-2018-11-13
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           227.32
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534073F8
BOARD_INFO_TYPE=main
```

> 
> > But where can I find what are the I2c commands for this device?
> 
> In the User's Guide from mindsensors.com.
> 
> [This page](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html) might also be helpful.

Thanks, but it's not obvious for me right now. I need to learn how to use direct i2c...

dlech, 2019-04-07:
Fix released in `ev3dev-2.3.3` kernel. The servo class in general seems a bit buggy still, but `rate_sp` should be working as expected now.

closed 2019-04-07 by dlech

cho934, 2019-04-09:
Could you remind me how to update this specific kernel ?
because I did'nt succeed by apt-get install linux-image-ev3dev-ev3 or by flash-kernel ev3-kernel-4.14.111-ev3dev-2.3.3-ev3 ?

dlech, 2019-04-09:
sudo apt update
    sudo apt upgrade

unless I botched something

cho934, 2019-04-09:
here the result of the update few minutes ago.. version 2.3.2...
```
...
Setting up systemd (232-25+deb9u11) ...
addgroup: The group `systemd-journal' already exists as a system group. Exiting.
Cannot set file attribute for '/var/log/journal', value=0x00800000, mask=0x00800000: Operation not supported
Cannot set file attribute for '/var/log/journal/a8a4655a020a4fa98e6be244f691f6d4', value=0x00800000, mask=0x00800000: Operation not supported
Setting up wget (1.18-5+deb9u3) ...
Setting up libnss-resolve:armel (232-25+deb9u11) ...
Processing triggers for dbus (1.10.26-0+deb9u1) ...
Setting up systemd-sysv (232-25+deb9u11) ...
Setting up libpam-systemd:armel (232-25+deb9u11) ...
Processing triggers for initramfs-tools (0.130) ...
update-initramfs: Generating /boot/initrd.img-4.14.96-ev3dev-2.3.2-ev3
I: The initramfs will attempt to resume from /dev/zram0
I: (UUID=1ef7bd61-2af6-4063-835b-faa7d1324ef5)
I: Set the RESUME variable to override this.
Using DTB: da850-lego-ev3.dtb
Alternate DTBs: da850-lego-ev3-lms2012.dtb da850-lego-ev3-adafruit18.dtb da850-lego-ev3-color.dtb
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-lms2012.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-lms2012.dtb
Taking backup of da850-lego-ev3-lms2012.dtb.
Installing new da850-lego-ev3-lms2012.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-adafruit18.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-adafruit18.dtb
Taking backup of da850-lego-ev3-adafruit18.dtb.
Installing new da850-lego-ev3-adafruit18.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-color.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-color.dtb
Taking backup of da850-lego-ev3-color.dtb.
Installing new da850-lego-ev3-color.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3.dtb
Taking backup of da850-lego-ev3.dtb.
Installing new da850-lego-ev3.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-lms2012.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-lms2012.dtb
Taking backup of da850-lego-ev3-lms2012.dtb.
Installing new da850-lego-ev3-lms2012.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-adafruit18.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-adafruit18.dtb
Taking backup of da850-lego-ev3-adafruit18.dtb.
Installing new da850-lego-ev3-adafruit18.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-color.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3-color.dtb
Taking backup of da850-lego-ev3-color.dtb.
Installing new da850-lego-ev3-color.dtb.
Installing /usr/lib/linux-image-4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3.dtb into /boot/dtbs/4.14.96-ev3dev-2.3.2-ev3/da850-lego-ev3.dtb
Taking backup of da850-lego-ev3.dtb.
Installing new da850-lego-ev3.dtb.
flash-kernel: installing version 4.14.96-ev3dev-2.3.2-ev3
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
Processing triggers for libc-bin (2.24-11+deb9u4) ...
robot@ev3dev:~$ 
```
And sudo apt-get dist-upgrade didn't add it too.

cho934, 2019-04-10:
And sudo apt-get dist-upgrade works now ;)

cho934, 2019-04-10:
set_rate_sp is working now, thanks a lot!