ev3dev/ev3dev #1243: How do I give an input to my ev3 brickman? I tried running the ev3 python file in my brick from my VScode but the input command appeared in the output window of  and I could not enter any values as output window is jus read-only. I Dont know how to reroute it to the TERMINAL WINDOW so that the python programs on the brick take in some input values like regular non-ev3 python programs

anzy0621, 2019-04-05:
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

anzy0621, 2019-04-05:
[image]

It says "cannot edit in read-only editor" Does ev3 python take in any input values? I searched everywhere but did not find any example ev3 python files that ask to input any values to the brickman like the non-ev3 python does using the "input"command

dlech, 2019-04-05:
To get a "real" terminal, you can open an SSH terminal in VS Code.

[image]

Then run your program like this:

    brickrun -r ./Python_ev3_codes/motor.py

The `-r` option redirects the input and output to the terminal like you want.

anzy0621, 2019-04-08:
Hi David thank you so much for helping me with that! But I have another issue while doing that. When I try to run the command you suggested above it says there’s no recognized “r” command. Then I ran brickrun —help to see the list of commands and the “ -r” command doesn’t seem to exist in the list of commands. Should I update something or add an extension? 
TIA 
Anusha

dlech, 2019-04-08:
It sounds like you don't have the most recent version. You can upgrade via `apt` or re-flash your microSD card.

anzy0621, 2019-04-08:
Thank you, David! It works,  I also, just have one last question! Sorry about that but really need help with it 
Does the input statement actually take in user input values in python ev3 or do I need a keyboard connected to brickman to enter values? Because when I run the program which has this statement 
x = input("enter a value") 

The ssh terminal takes in an integer value that I enter but does not do anything with it.. the program keeps running forever until I press the middle button on the brickman then the program just runs but doesn't do anything with the integer input value I just gave in. Is it possible that its something not yet included in the ev3 python yet? I'm trying to give in some integer values that the brickman takes and passes this value as an argument to to sensors/motors as distance or speed etc.
[image]

dlech, 2019-04-08:
I don't see where you used the `-r` option. It won't work without it.

anzy0621, 2019-04-08:
Sorry, I got confused, I thought the brick run --directory was the same thing. I have ev3 stretch 2018-11-13 version and I cannot find any other latest version than this. And my sudo apt commands don't work as shown below so I don't know how else to upgrade/update
[image]
. Is it because I'm using vscode?

dlech, 2019-04-08:
You have to connect your EV3 to the Internet to upgrade packages. https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

anzy0621, 2019-04-08:
Thank you David! My issue is resolved! Thank you so much for your assistance, it was of great help to me! My brick is taking user input values now and running the motors!

dlech, 2019-04-08:
Glad to hear it is working.

closed 2019-04-08 by dlech