ev3dev/ev3dev #1244: putfont: KDFONTOP: Operation not permitted

dlech, 2019-04-05:
Same issue with ev3dev stretch

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-11-13
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           227.32
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534073F8
BOARD_INFO_TYPE=main
```

Here is the error:
```
robot@ev3dev:~$ setfont /usr/share/consolefonts/Lat7-Terminus12x6.psf.gz 
putfont: KDFONTOP: Operation not permitted
robot@ev3dev:~$ 
```

_Originally posted by @cho934 in https://github.com/ev3dev/ev3dev/issues/764#issuecomment-480446599_

dlech, 2019-04-05:
Are you running this on a remote terminal? It requires a *virtual console*, which means it can't run remotely.

You can kind of hack around this by running `sudo chvt 6` to force the display on the EV3 to switch to a login prompt, then `sudo conspy` to mirror that terminal locally.

cho934, 2019-04-05:
I'm using the ssh console using wifi connection.

closed 2019-05-26 by dlech