ev3dev/ev3dev #1247: mindsensors.com 8-channel Servo Controller - system_error sometimes

cho934, 2019-04-10:
Hello,

Sometimes with the same program, I received an error in my unittests for servos.
I had a some additional infos in the ev3dev.cpp for exception.
So in this example, pos was put to 0 after the -100 position, and throw an exception of the path which is the good one (and worked 4 times just  before)

robot@ev3dev:~/pmx$ ./Driver-UnitTest_EV3 
ServoDriver.EV3 DEBUG ServoDriver()
ServoDriverTest.EV3 DEBUG testServos()
ServoDriverTest.EV3 INFO pos = 0
ServoDriverTest.EV3 INFO pos = 100
ServoDriverTest.EV3 INFO pos = 0
ServoDriverTest.EV3 INFO pos = -100
terminate called after throwing an instance of 'std::system_error'
  what():   [set_attr_int] value=0 path=/sys/class/servo-motor/motor16/position_sp: No such device or address
Aborted

In your opinion, what could be wrong ? timing ?

dlech, 2019-04-10:
If you run the same test multiple times, does it always fail? In the same place?

cho934, 2019-04-11:
No, because, next runs are ok. And it is random, sometimes I can't write and it's not the same...
I'm going to log the "attempt", in the cpp code, the function try normally 2 times...

cho934, 2019-04-11:
Maybe this issue is linked
https://github.com/ddemidov/ev3dev-lang-cpp/issues/47

dlech, 2019-04-11:
The "No such device or address" is probably coming from the I2C layer in the kernel. And yes, is probably a timing issue with the servo controller. The servo controller also shows up as a lego-sensor device and lets you read the battery voltage. So there are background polls on the I2C bus while your program is running. You can use the `poll_ms` attribute to disable polling by setting it to 0.

Besides that, it is probably best to just add a retry to your program when you get ENODEV.

dlech, 2019-04-13:
I just got an ENXIO (`nxt-i2c-sensor 6-0058: Error registering servos. -6`) while probing the sensor, so this is definitely a hardware issue. Ideally it would be fixed in the mindsensors firmware, but if we can figure out a retry scheme that works, we could probably implement that in the kernel.

cho934, 2019-07-05:
I'm interesting to use servos without have a crash during a match ;)