ev3dev/ev3dev #1248: brickpi3: motor stays in running state when number (vs string) is written to position_sp

dwalton76, 2019-04-12:
**Describe the bug**
If we write a number to position_sp the motor sticks in `running` for forever but if we write the string of the same number it is fine.

After running this:
```
robot@brickpi3:~/ev3dev-lang-python$ cat ~/foo.sh 
#!/bin/bash

echo "reset" > /sys/class/tacho-motor/motor0/command
echo 907 > /sys/class/tacho-motor/motor0/position_sp
echo 480 > /sys/class/tacho-motor/motor0/speed_sp
echo "hold" > /sys/class/tacho-motor/motor0/stop_action
echo "run-to-rel-pos" > /sys/class/tacho-motor/motor0/command
robot@brickpi3:~/ev3dev-lang-python$ 
```

we see
```
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/state
running
robot@brickpi3:~/ev3dev-lang-python$ 
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
908
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position_sp
907
robot@brickpi3:~/ev3dev-lang-python$ 
```
It always seems to end up at 908 instead of 907

but after running the following where we write the string 907 instead of the number 907
```
robot@brickpi3:~/ev3dev-lang-python$ cat ~/foo2.sh 
#!/bin/bash

echo "reset" > /sys/class/tacho-motor/motor0/command
echo "907" > /sys/class/tacho-motor/motor0/position_sp
echo 480 > /sys/class/tacho-motor/motor0/speed_sp
echo "hold" > /sys/class/tacho-motor/motor0/stop_action
echo "run-to-rel-pos" > /sys/class/tacho-motor/motor0/command
robot@brickpi3:~/ev3dev-lang-python$ 
```

we see
```
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/state
holding
robot@brickpi3:~/ev3dev-lang-python$ 
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position
907
robot@brickpi3:~/ev3dev-lang-python$ cat /sys/class/tacho-motor/motor0/position_sp
907
robot@brickpi3:~/ev3dev-lang-python$ 
```

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2019-03-03
Kernel version:     4.14.98-ev3dev-2.3.3-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dlech, 2019-04-12:
That's weird. I don't think the quotes should matter in bash.

dlech, 2019-04-13:
> It always seems to end up at 908 instead of 907

I think getting within 1 degree is pretty good. And if you want better, you'll have to ask Dexter Industries for a firmware update. :wink: 

> If we write a number to position_sp the motor sticks in `running` for forever but if we write the string of the same number it is fine.

I think the difference is just running it a second time, not the quotes.

dlech, 2019-04-13:
So would you rather have this bug  or #1237? Basically, we are relying on the BrickPi3 firmware to tell us what the motor is doing, but it isn't 100% compatible with the way ev3dev handles motors.

dwalton76, 2019-04-14:
I could live with #1237.  For that would could the delta still be tightened up some from 10 degrees?

dlech, 2019-04-14:
If anything, it needs to be bigger than 10 degrees, not smaller. But I was able to find a different workaround for #1237 that should make things better there without the side effect of this issue.

dlech, 2019-05-26:
I think this was resolved in the ev3dev-2.3.3 kernel.

closed 2019-05-26 by dlech