ev3dev/ev3dev #1249: UART sensors broken in ev3dev-2.3.3 kernel

dlech, 2019-04-13:
**Describe the bug**
After upgrading to `4.14.111-ev3dev-2.3.3-ev3` kernel, EV3 UART sensors are no longer detected.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade kernel to `4.14.111-ev3dev-2.3.3-ev3`
2. Reboot
3. Plug in a UART sensor
4. No sensor appears

**Expected behavior**
Sensor should work like previous versions

**Robot (please complete the following information):**

```
Image file:         ev3dev-stretch-ev3-generic-2018-06-13
Kernel version:     4.14.111-ev3dev-2.3.3-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533F4838
BOARD_INFO_TYPE=main

```

**Additional context**

`dmesg` shows...

```
[    2.228164] lego-port port1: Added new device 'ev3-ports:in3:ev3-uart-host'
[   51.547561] Registered EV3 UART sensor line discipline. (29)
```

so we are getting part of the way there.

Also reported by `CocoaGeek:` on IRC.

dlech, 2019-04-13:
Regression probably due to https://github.com/ev3dev/lego-linux-drivers/commit/2ce65b084c42c251abaa3ac8f86bd57cb553f02e

DJuego, 2019-04-21:
I think I have this problem! :-O

DJuego

Chickenboy445, 2019-04-22:
How can I have my sensors working? Should I download an older version of the kernel? If so, which one should I pick?

JorgePe, 2019-04-22:
ev3dev released with LEGO Education EV3 MicroPython seems to works fine:

```
Image file:         ev3-micropython-v1.0.0-sd-card-image
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
```
the kernel is from Jan 27 so is recent

dlech, 2019-04-22:
Try this:

    sudo flash-kernel 4.14.96-ev3dev-2.3.2-ev3
    sudo reboot

Chickenboy445, 2019-04-22:
That worked! Thank you so much for your help. I have a pretty tough project to do with this and without sensors it would have been impossible. Thanks!

sdebaerd, 2019-05-04:
I'm experiencing the same problem, but it's not resolved after flashing the older kernel.
The sensor page remains empty, also when trying to use the java lang, the jar crashes with the following stack:

Exception in thread "main" java.lang.RuntimeException: The path doesn't exist: /sys/class/lego-sensor
	at ev3dev.utils.Sysfs.getElements(Sysfs.java:107)
	at ev3dev.hardware.EV3DevDevice.detect(EV3DevDevice.java:57)
	at ev3dev.hardware.EV3DevSensorDevice.<init>(EV3DevSensorDevice.java:40)
	at ev3dev.sensors.BaseSensor.<init>(BaseSensor.java:22)
	at ev3dev.sensors.ev3.EV3ColorSensor.<init>(EV3ColorSensor.java:48)
	at example.programming.SimpleWebServer.<init>(SimpleWebServer.java:41)
	at example.programming.SimpleWebServer.main(SimpleWebServer.java:166)

When using the "old" lejos image or standard mindstorms firmware, the sensor is recognized.

salamanders, 2019-05-07:
+1, new install (new user!)

```
Exception in thread "main" java.lang.RuntimeException: The path doesn't exist: /sys/class/lego-sensor
	at ev3dev.utils.Sysfs.getElements(Sysfs.java:107)
	at ev3dev.hardware.EV3DevDevice.detect(EV3DevDevice.java:57)
	at ev3dev.hardware.EV3DevSensorDevice.<init>(EV3DevSensorDevice.java:40)
	at ev3dev.sensors.BaseSensor.<init>(BaseSensor.java:22)
	at ev3dev.sensors.ev3.EV3UltrasonicSensor.<init>(EV3UltrasonicSensor.java:58)
```

dlech, 2019-05-07:
workaround: https://github.com/ev3dev/ev3dev/issues/1249#issuecomment-485442515

sdebaerd, 2019-05-07:
Hi

For me the workaround is not fully working, as the registration of the uart is not appearing in the dmesg, also the sensor screen remains blank. I have followed all steps in the workaround, but still no joy. 

If you need more information, please let me know. 

Thanks in advance.

dlech, 2019-05-07:
Can you confirm the kernel version that is actually running on your EV3?

sdebaerd, 2019-05-07:
I've executed the workaround you provided, but I will get the details by Thursday, as I'm currently abroad.

salamanders, 2019-05-08:
trying `sudo flash-kernel 4.14.96-ev3dev-2.3.2-ev3`

Nope.  Odd.  I still get `Exception in thread "main" java.lang.RuntimeException: The path doesn't exist: /sys/class/lego-sensor	at ev3dev.utils.Sysfs.getElements(Sysfs.java:107)`

But the path does exist.  

```
robot@ev3dev:~$ ls -la /sys/class/lego*
/sys/class/lego-port:
total 0
drwxr-xr-x  2 root root 0 May  8 00:56 .
drwxr-xr-x 48 root root 0 May  8 00:56 ..
lrwxrwxrwx  1 root root 0 May  8 00:56 port0 -> ../../devices/platform/ev3-ports/ev3-ports:in1/lego-port/port0
lrwxrwxrwx  1 root root 0 May  8 00:56 port1 -> ../../devices/platform/ev3-ports/ev3-ports:in2/lego-port/port1
lrwxrwxrwx  1 root root 0 May  8 00:56 port2 -> ../../devices/platform/ev3-ports/ev3-ports:in3/lego-port/port2
lrwxrwxrwx  1 root root 0 May  8 00:56 port3 -> ../../devices/platform/ev3-ports/ev3-ports:in4/lego-port/port3
lrwxrwxrwx  1 root root 0 May  8 00:56 port4 -> ../../devices/platform/ev3-ports/ev3-ports:outA/lego-port/port4
lrwxrwxrwx  1 root root 0 May  8 00:56 port5 -> ../../devices/platform/ev3-ports/ev3-ports:outB/lego-port/port5
lrwxrwxrwx  1 root root 0 May  8 00:56 port6 -> ../../devices/platform/ev3-ports/ev3-ports:outC/lego-port/port6
lrwxrwxrwx  1 root root 0 May  8 00:56 port7 -> ../../devices/platform/ev3-ports/ev3-ports:outD/lego-port/port7

/sys/class/lego-sensor:
total 0
drwxr-xr-x  2 root root 0 May  8 00:56 .
drwxr-xr-x 48 root root 0 May  8 00:56 ..
```

dlech, 2019-05-08:
If `uname -r` shows the correct kernel version, then you should bring up the issue in the ev3dev-lang-java project.

salamanders, 2019-05-08:
Good call - it still shows 4.14.111-ev3dev-2.3.3-ev3.
Happy to blow the whole thing away if it would be easier to reflash from the older version...
Oh hey.  Need `--force`
ooos.  Stuff showing up in lego-sensor.  This is good.

sdebaerd, 2019-05-08:
Makes sense, since we need to flash a lower version, forgot about this option. Can't wait to to get back home to test it. I will let you know the outcome.

salamanders, 2019-05-08:
My sensors work, this workaround fixed it for me, thank you!

sdebaerd, 2019-05-09:
Got home at last. When I try to flash the old version, with --force, it keeps mentioning that the 111 is the newer release and it's ignoring it, when using the following command:
sudo flash-kernel 4.14.96-ev3dev-2.3.2-ev3 --force
sudo reboot

I then moved the --force flag and it started to deploy the old kernel:
sudo flash-kernel --force 4.14.96-ev3dev-2.3.2-ev3
sudo reboot

dlech, 2019-05-09:
Fix released in ev3dev-2.3.4 kernel

closed 2019-05-09 by dlech