ev3dev/ev3dev #1251: Option to change hostname in Brickman

JorgePe, 2019-04-16:
Now that more people will use Visual Studio Code with ev3dev  thanks to LEGO Education new Ev3 MicroPython image, chances of using Wi-Fi and more than 1 brick on the same room will raise.

Talking with a LEGO engineer a few days ago about the new SPIKE not having Wi-Fi I realised that he considered Wi-Fi as a messy feature in (standard) EV3 classrooms.

So an easy option to change the brick's hostname would be nice - as long as both Wi-Fi and the user's computer support Avahi/Bonjour/mDNS connection would be simple.

We can change the hostname from a SSH session but I don't see an option in Brickman to do it.

devinbreise, 2020-04-09:
+1 on this feature.  In addition, many/most classrooms will be using Bluetooth with their EV3s (instead of adding a wifi dongle if they use wireless code downloading.  Ideally, this feature would work for that network interface as well.

snaildos, 2021-03-25:
Still waiting for this, this is require to do.