ev3dev/ev3dev #1252: Build stretch image from scratch?

JunOllyLi, 2019-04-17:
I have followed the instructions to setup the Dockers image and now I have the cross-compiler.
But is there any document describing how to create the SD image from source code?

Thanks!

dlech, 2019-04-17:
We don't create the SD card image from source code, we create it from already compiled Debian packages.

We use the [docker-library](https://github.com/ev3dev/docker-library) to create a Docker image, then use [brickstrap](https://github.com/ev3dev/brickstrap) to turn the Docker image into an SD card image.

JunOllyLi, 2019-04-17:
@dlech,
Thanks for the reply. But is there any document describing how to build the docker image?
I am new to the Debian stuff and Docker, although I had plenty of experience with other build environment such as Yocto.
I was thinking something like Yocto, you get a list of recipes from some repository and just do "bitbake ..." and it pull sources for each component from Internet and ends up with your result image.
I am sure you guys should have some kind of build scripts or at least instructions for manual steps, right? It would be hard to imagine that every one has to memorize all steps and do it manually.

dlech, 2019-04-17:
Check out the `.travis.yml` file in both of the repositories that I linked. It shows how the automated builds work.

dlech, 2019-04-17:
If you don't want to modify the image, you can skip the `docker-library` part.

JunOllyLi, 2019-04-18:
Thanks a lot! I will give it a try!

closed 2019-04-24 by JunOllyLi