ev3dev/ev3dev #1253: EV3 sometimes doesn't see motor attached.

artpoz, 2019-04-17:
**Describe the bug**
EV3 sometimes doesn't see motor attached.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Device Browser"
2. Go to 'Motors'
3. Unplug and plug Motor between different ports (for example D and C)
Result
You don't see motor.
```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```
Image file:         ev3-micropython-v1.0.0-sd-card-image
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:          2.1
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165343C8E0
BOARD_INFO_TYPE=main
```

dlech, 2019-04-22:
I am not able to reproduce the problem. Can you give more details? For example, what is plugged into each port? Are you using any USB devices? Are there any extra programs running in the background? 

When this problem happens and the motor is plugged in, what is the output of

    for f in /sys/bus/iio/devices/iio\:device0/*_raw; do echo $(basename $f); cat $f; done

dlech, 2019-04-22:
for reference: https://www.ev3dev.org/docs/kernel-hackers-notebook/ev3-adc/

artpoz, 2019-04-22:
Hardware:
Large Motor pluged in A
Large Motor pluged in B
Nothing in USB
Nothing run.

Brickman->Motors:
lego-ev4-lmotor at ev3-ports:outA

Output
in_voltage0_raw
916
in_voltage10_raw
1022
in_voltage11_raw
11
in_voltage12_raw
1022
in_voltage13_raw
510
in_voltage14_raw
512
in_voltage15_raw
0
in_voltage1_raw
920
in_voltage2_raw
0
in_voltage3_raw
24
in_voltage4_raw
750
in_voltage5_raw
12
in_voltage6_raw
1022
in_voltage7_raw
12
in_voltage8_raw
1022
in_voltage9_raw
11

Brickman doesn't see motor B.

artpoz, 2019-04-22:
After more tests it turned out that one cable is broken. Problem fixed. I'm closing this issue.

closed 2019-04-22 by artpoz