ev3dev/ev3dev #1254: Boot speed of ev3dev-stretch

xdsl, 2019-04-19:
How to speed up of boot ev3dev-stretch on EV3?
ev3dev-stretch ~ 1min 30sec
ev3dev-jessie ~ 50sec

dlech, 2019-04-19:
There are tools like `systemd-chart` that can be used to profile the boot. Also, searching the web for "embedded linux boot time" will come up with lots of articles. It is a bit late to do much about ev3dev-stretch though.

staadecker, 2019-04-20:
My brick took almost 3minutes to boot. Here's the output from ` systemd-analyze plot`. Any ideas to reduce the time? Is it the SD card?

`connman-wait-online.service` is what took the longest (1min 41s). Any way to disable it?

[image]

dlech, 2019-04-20:
Although `connamn-wait-online.service` takes a very long time, it is not actually blocking anything. One we get to `brickman.service`, we are basically done booting. So really, the longest thing is the kernel itself.

You can disable services with `sudo systemctl disable x.service` or mask them with `sudo systemctl mask x.service` if they are being really stubborn and running anyway. You could try disabling non-essential services. I would suggest `keyboard-setup.service`, `fake-hwclock.service`, `systemd-journald.service`, `systemd-fsck-root.service`, `systemd-journal-flush.service`, `networking.service`, `console-setup.service`, `nfs-config.service`.

dlech, 2019-04-20:
FWIW, [this](https://events17.linuxfoundation.org/sites/events/files/slides/12%20Lessons%20Learnt%20in%20Boot%20Time%20Reduction.pdf) is one of the best talks I've seen on the subject. There is a video that goes with it somewhere (can't find it now).

staadecker, 2019-04-21:
Ok. So I've disabled the services as mentioned. Now the boot took over 2 minutes for brickman to start. It seems `systemd-fsck-root.service` is taking up a lot of time (59s to start). Any suggestions? Here is the plot.

[image]

dwalton76, 2019-04-21:
Mine went from 2m 15s to 1m 51s

dlech, 2019-04-22:
The type of SD card you are using probably also has a significant impact. https://github.com/ev3dev/ev3dev/wiki/Selecting-a-microSD-card

dlech, 2019-04-22:
> It seems `systemd-fsck-root.service`

What happens if you disable or mask that one?

xdsl, 2019-04-23:
Boot time:
ev3dev-jessie kernel - 3-5 sec
ev3dev-stretch kernel - 15-20 sec 
this is normal?

staadecker, 2019-05-10:
Sorry for the slow reply. After masking the service I get this. Brickman boots after 1 min.

[image]

dlech, 2019-10-08:
Interesting thread about BeagleBone boot time.

https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/beagleboard/KRiyTWEqFCg/SxwBnVnvAgAJ