ev3dev/ev3dev #1255: brickman build failure `standard_init_linux.go:207: exec user process caused "exec format error"`

dwalton76, 2019-04-19:
The README says to do `./docker/setup.sh $BUILD_AREA $ARCH` but from reading setup.sh it should just be `./docker/setup.sh $ARCH`.  Anyway...the docker container is failing to start due to a `standard_init_linux.go:207: exec user process caused "exec format error"` error:

```
dwalton@laptop:brickman$ echo $ARCH
armel
dwalton@laptop:brickman$ ./docker/setup.sh $ARCH
Sending build context to Docker daemon   5.12kB
Step 1/2 : FROM ev3dev/debian-stretch-armel-cross
 ---> caef6e789297
Step 2/2 : RUN sudo apt-get update &&     DEBIAN_FRONTEND=noninteractive sudo apt-get install --yes --no-install-recommends         cmake         libev3devkit-dev         libgirepository1.0-dev         libgudev-1.0-dev         netpbm         valac
 ---> Running in 0e8cad4eb6b6
standard_init_linux.go:207: exec user process caused "exec format error"
The command '/bin/sh -c sudo apt-get update &&     DEBIAN_FRONTEND=noninteractive sudo apt-get install --yes --no-install-recommends         cmake         libev3devkit-dev         libgirepository1.0-dev         libgudev-1.0-dev         netpbm         valac' returned a non-zero code: 1
dwalton@laptop:brickman$ 
```

Trying to start it straight from the command line fails with the same error
```
dwalton@laptop:brickman$ 
dwalton@laptop:brickman$ docker image ls ev3dev/debian-stretch-armel-cross
REPOSITORY                          TAG                 IMAGE ID            CREATED             SIZE
ev3dev/debian-stretch-armel-cross   latest              caef6e789297        13 months ago       829MB
dwalton@laptop:brickman$ 
dwalton@laptop:brickman$ docker run ev3dev/debian-stretch-armel-cross
standard_init_linux.go:207: exec user process caused "exec format error"
dwalton@laptop:brickman$ 
```

docker --version
```
dwalton@laptop:brickman$ docker --version
Docker version 18.09.4, build d14af54266
dwalton@laptop:brickman$ 
```

dlech, 2019-04-19:
You probably need to install `qemu-user-static`.

dwalton76, 2019-04-19:
Yep that was it, thanks!  Will submit a PR to update the README.

closed 2019-04-19 by dwalton76