ev3dev/ev3dev #1256: brickman build: where to find libev3devkit-dev and grx

dwalton76, 2019-04-19:
I installed all of the other packages in `debian/control` but could not find `libev3devkit-dev`, tried to build anyway to see how far it would get:
```
dwalton@laptop:brickman-desktop-on-laptop$ cmake ~/lego/brickman/ -DCMAKE_BUILD_TYPE=string:Debug -DBRICKMAN_TEST=bool:Yes
-- Checking for modules 'ev3devkit-0.5;glib-2.0;gobject-2.0;gmodule-2.0;gio-2.0;gio-unix-2.0;grx-3.0;gudev-1.0'
--   No package 'ev3devkit-0.5' found
--   No package 'grx-3.0' found
```

Where do I find `libev3devkit-dev` and `grx`?
```
dwalton@laptop:brickman-desktop-on-laptop$ apt-cache search ev3dev
dwalton@laptop:brickman-desktop-on-laptop$  
dwalton@laptop:brickman-desktop-on-laptop$ apt-cache search grx
dwalton@laptop:brickman-desktop-on-laptop$ 
```

dlech, 2019-04-20:
You should be able to get grx from the ev3dev-tools PPA. But it doesn't look like ev3devKit is there.

What I would really like to do instead though is make use of the [CMake export](https://cmake.org/cmake/help/v3.7/command/export.html) feature so that you can build all 3 from source easily at the same time.

dwalton76, 2019-04-21:
thanks, was able to install `libgrx-3.0-dev`.  I see where `libev3devkit-dev` is already in the docker container that is used for building brickman...where did it come from to be placed in the docker container?

dlech, 2019-04-26:
The Docker container is ev3dev-stretch with armel (or armhf) architecture. So it pulls the exact same packages that are installed on the EV3. To build on desktop, you need a version of the package built for the distro/architecture of your desktop.

dlech, 2019-09-04:
The cmake export stuff is done now, so it is easier to work on the full stack (grx, ev3devkit, brickman)