ev3dev/ev3dev #1257: Sudo APT get update command does not work

anzy0621, 2019-04-20:
Hello,
           I'm using python ev3 with visual studio code. I wanted to use an SSH terminal to give in user input values to my python programs. However, in order to give input through SSH terminal I needed to have the "brickrun - r" command. So I connected my ev3 brick to the internet first following this tutorial https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/ and my ev3 brick says "Staus: Connected". However when I run the sudo apt get update command it says failed to resolve host and cannot connect to the internet to grab updates from the respective websites. I tried different wifi networks, restarting brick, disconnecting and reconnecting internet connection. I don't know what else to to, please help. It's weird how it says on the brick that the STATUS IS CONNECTED but ev3 is not able to use internet when i use sudo apt get update

staadecker, 2019-04-20:
Same issue as #1154

dlech, 2019-04-22:
Which tutorial did you follow to connect to the Internet? What OS does your computer run?