ev3dev/ev3dev #1258: Root Access in File Explorer

MrMan999, 2019-04-20:
Is it possible to access /root in File Explorer
I can already access it in Terminal with ssh

dlech, 2019-04-20:
Why do you need root access?

MrMan999, 2019-04-21:
> Why do you need root access?

I wanted to know because I was interested to see its system files and see the full directory

Also, I tried typing: 
`root@ev3dev:~# sudo usermod -d / robot`

but then it said that the user was already logged in:
`usermod: user robot is currently used by process 373`

so is it possible to switch the user to root user on the robot with a startup script?

dlech, 2019-04-22:
Thanks for the explanation. It is not possible to browse directories above `/home/robot` in Brickman. And having root access won't make a difference.

I don't really recommend it, but if you really want to do this,you can make a symlink to the root folder from the `/home/robot` folder. In an SSH terminal:

    cd /home/robot
    mkdir sysroot
    ln -s /* sysroot/

However, there is a `tree` command you can use to see the same thing from a terminal.

    tree --help  # see all the options
    tree /bin | less  # list everything under /bin
    tree -C /bin | less -r  # same thing with pretty colors
    tree -L 1 /  # only go one level deep

MrMan999, 2019-04-22:
Thanks, I'll close this question

MrMan999, 2019-04-22:
Wait, I can't access it, the files/folders are all executables, when I opened the .err.log files, it said I don't have permission
I tried using : `sudo chmod -R 777 /`
and tried restarting it, but it didn't work
please help

dlech, 2019-04-22:
You can't open text files with Brickman.

Also you should re-flash your SD card to fix all of the file permissions.

MrMan999, 2019-04-22:
I meant that everything in sysroot is an executable
but if I really need to, I would reflash the SD card
Will reflashing it delete all my projects?

dlech, 2019-04-22:
> Will reflashing it delete all my projects?

yes

MrMan999, 2019-04-23:
Where are the projects stored in the sd card?

JorgePe, 2019-04-23:
If you are using default user ("robot") they are under "/home/robot/"

MrMan999, 2019-05-04:
I mean in the sd card, I can't seem to find any linux directories in the SD card

MrMan999, 2019-05-04:
I found the boot.scr file, uEnv.txt file, u-boot.bin, some android files like .android_secure, and a bunch of FSCK####.REC files

should I delete these FSCK####.REC files