ev3dev/ev3dev #1262: GyroSensor "Reconnected due to: No data since last keep-alive."

dwalton76, 2019-04-21:
Running
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165346201F
BOARD_INFO_TYPE=main
```

I plugged the gyro in and dmesg showed
```
[Apr21 20:26] lego-port port0: Added new device 'ev3-ports:in1:ev3-uart-host'
[  +1.142048] Registered EV3 UART sensor line discipline. (29)
[  +2.583202] lego-sensor sensor0: Registered 'lego-ev3-gyro' on 'ev3-ports:in1'.
```

Next I ran the following to print the gyro angle every 1s
```
robot@ev3dev:~$ cat test-gyro.py 
#!/usr/bin/env python3

from ev3dev2.sensor.lego import GyroSensor
from time import sleep
gyro = GyroSensor()
gyro.reset()

while True:
    print(gyro.angle)
    sleep(1)
robot@ev3dev:~$ 
```
 
It normally gets about 4 or 5 passes through the loop and then the gryo reconnects
```
Reconnected due to: No data since last keep-alive.
```

I am seeing this on an old (03N3) gryo and a new (22N5).  Sometimes the kernel locks up...not sure what the trigger is for when it does vs does not.

dlech, 2019-04-22:
Possibly related to #671

dlech, 2019-04-22:
The sample code doesn't seem to work.

```
Traceback (most recent call last):
  File "./issue1262", line 6, in <module>
    gyro.reset()
  File "/usr/lib/python3/dist-packages/ev3dev2/sensor/lego.py", line 627, in reset
    self._direct = self.set_attr_raw(self._direct, 'direct', 17)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 306, in set_attr_raw
    return self._set_attribute(attribute, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 264, in _set_attribute
    self._raise_friendly_access_error(ex, name)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 269, in _raise_friendly_access_error
    raise driver_error
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 261, in _set_attribute
    attribute.write(value)
TypeError: a bytes-like object is required, not 'int'
```

```
$ python3 -c 'import ev3dev2; print(ev3dev2.__version__)'
2.0.0beta3
```

dlech, 2019-04-22:
I commented out the `reset()` line and I cannot reproduce the problem.

dwalton76, 2019-04-22:
I think you need this patch:
https://github.com/ev3dev/ev3dev-lang-python/commit/195259f1c54b1fecf08257e762f748406acd2538

dlech, 2019-04-22:
So the problem only happens when `gyro.reset()` is part of the program? If you comment it out, the problem goes away?

dwalton76, 2019-04-22:
I'll can test tonight when I get home

dwalton76, 2019-04-23:
Confirmed that the problem only happens after calling `gyro.reset()`

dwalton76, 2019-04-26:
@dlech do you need anything else on this one?  I can reproduce it everytime if I call `gyro.reset()`

dlech, 2019-11-24:
This is an ev3dev-lang-python bug. https://github.com/ev3dev/ev3dev-lang-python/pull/692

closed 2019-11-24 by dlech