ev3dev/ev3dev #1263: Can't import requests module

Arsik0153, 2019-04-22:
I want to execute this program:
```
#!/usr/bin/env pybricks-micropython

from pybricks import ev3brick as brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import (Port, Stop, Direction, Button, Color,
                                 SoundFile, ImageFile, Align)
from pybricks.tools import print, wait, StopWatch
from pybricks.robotics import DriveBase
import requests

# Write your program here
brick.sound.beep()
response = requests.get('https://httpbin.org/basic-auth/user/passwd', auth=('user', 'passwd'))
print(response.content)
```

but I have error: 
```
Started.
----------
Traceback (most recent call last):
  File "/home/robot/sdfsdf/main.py", line 10, in <module>
ImportError: no module named 'requests'
----------
Exited with error code 1.

```

Trying to install this module: 
```
robot@ev3dev:~$ sudo apt-get install python3-requests
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-requests is already the newest version (2.12.4-1).
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
```

BUT, when I execute it in terminal, I have no error:
```
robot@ev3dev:~$ sudo python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> response = urllib.request.urlopen('https://httpbin.org/get')
>>> print(response.read())
b'{\n  "args": {}, \n  "headers": {\n    "Accept-Encoding": "identity", \n    "Host": "httpbin.org", \n    "User-Agent": "Python-urllib/3.5"\n  }, \n  "origin": "37.151.19.22, 37.151.19.22", \n  "url": "https://httpbin.org/get"\n}\n'
```
What I should do?

JorgePe, 2019-04-22:
You are mixing pythons.
You are using LEGO EV3 MicroPython image so officially this is not related to ev3dev.

'pybricks-micropython' is the micropython that LEGo reelased. It's a custom implementation of MicroPython for the EV3 that runs on ev3dev but it is  not the same has 'python3'.

micropython requires it's own modules, I think that on LEGO image you only have those available with ''micropython-lib, please see
https://github.com/micropython/micropython-lib

there is urequests on that list, perhaps it works (not all on that list work, some are dummy libraries still wainting to be implemented)

JorgePe, 2019-04-22:
urequest work but doesn't seem complete, you will probably have to check other ways:

```
robot@iota:~$ micropython 
MicroPython v1.9.4 on 2018-05-22; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import urequests
>>> response = urequests.get('https://httpbin.org/basic-auth/user/passwd', auth=('user', 'passwd'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests/urequests.py", line 112, in get
TypeError: unexpected keyword argument 'auth'
>>> response = urequests.get('https://google.com/')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests/urequests.py", line 112, in get
  File "urequests/urequests.py", line 100, in request
  File "urequests/urequests.py", line 97, in request
NotImplementedError: Redirects not yet supported
```

dlech, 2019-04-22:
You should be able to manually create the authentication header and use `headers=` instead of using `auth=`.

https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Authorization

Agrobots, 2021-04-11:
!

#!/usr/bin/env pybricks-micropython 

!