ev3dev/ev3dev #1264: Internet over USB on Ubutnu 18.04

ifarup, 2019-04-23:
I have a similar problem as in #1109 with internet over USB with Ubutnu 18.04. I tried both jessie-ev3-generic-2017-09-14 and stretch-ev3-generic-2019-03-03 and followed the online guides closely (using nm-connection-editor to access the shared network): https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/  The connection seems to come up. The EV3 reports the IP address (e.g. 169.254.120.209) and the state in the display goes to "State: Connected" (not "Online" as it should be according to the online docs). And the computer seems to know about it, but not being able to reach it:

$ ping ev3dev.local
PING ev3dev.local (169.254.120.209) 56(84) bytes of data.
From x1 (192.168.122.1) icmp_seq=1 Destination Host Unreachable

$ ssh robot@ev3dev.local
ssh: connect to host ev3dev.local port 22: No route to host

Is there something more I need to do with my network settings?

dlech, 2019-04-23:
Can you post screenshots of your network configuration?

ifarup, 2019-04-23:
Sure, will these do?

[image]
[image]
[image]

dlech, 2019-04-23:
Thanks. I looks like everything is configured correctly.

I wonder if you are missing one of the [recommended packages](https://packages.ubuntu.com/bionic/network-manager) for NetworkManager like `dnsmasq-base` or `iptables`. 

Also, if you haven't already, rebooting everything sometimes magically fixes problems.

ifarup, 2019-04-24:
I have tried rebooting both the laptop and the ev3 several times, and also to plug the USB cable before and after rebooting in various sequences. All the recommended packages are installed. Actually, all the listed packages but libteam-utils were installed, so I tried installing that one and rebooting everything again. Unfortunately, the behaviour stays the same.

ifarup, 2019-04-24:
But is this correct? It seems that both the EV3 and pan1 come up as ip4 10.42.0.1, whereas the ev3 reports the ip 169.254.X.X:

$ ifconfig 
enp0s20f0u2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.0.1  netmask 255.255.255.0  broadcast 10.42.0.255
        inet6 fe80::21c5:66a8:d362:c4c7  prefixlen 64  scopeid 0x20<link>
        ether 12:16:53:65:d6:4e  txqueuelen 1000  (Ethernet)
        RX packets 81  bytes 18983 (18.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 59  bytes 6943 (6.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[...]

pan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.0.1  netmask 255.255.255.0  broadcast 10.42.0.255
        inet6 fe80::580f:c1ff:feb9:31b  prefixlen 64  scopeid 0x20<link>
        ether 5a:0f:c1:b9:03:1b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 90  bytes 9517 (9.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ifarup, 2019-04-24:
With help from a good colleague, we found the problem and two solutions. Some other program must have made the pan1 bridge interface which collides with the ev3dev interface, so dnsmasq is not coming up. Removing the pan1 interface (which reappears on reboot) or assigning a different ip address to the ev3dev interface, everything works smoothly!

[image]

closed 2019-04-24 by ifarup

ifarup, 2019-04-24:
It turns out that the conflicting pan1 bridge interface is actually made by blueman that I installed following the instructions here:

https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

Removing blueman and the pan1 interface, everything works according to your descriptions. You might want to make a note about this conflict between USB and Bluetooth on your tutorials.

dlech, 2019-04-24:
Thanks for digging into this!