ev3dev/ev3dev #1267: Communicating over Bluetooth between 2 EV3 bricks with UART

runningOutOfUsernameIdeas, 2019-04-25:
Hello there!

My Goal: Have the ability to send bytes of data via a bluetooth connection bettween 2 ev3 bricks using UART

Backstory: I recently found this webpage: https://www.ev3dev.org/docs/kernel-hackers-notebook/ev3-uart/ which stated this:
'
Port | Use
-- | --
UART0 | Input port 2 UART sensors.
UART1 | Input port 1 UART sensors. Also debugging terminal (starts at boot).
UART2 | Bluetooth.
PRU0/SUART0 | Input port 4 UART sensors.
PRU0/SUART1 | Input port 3 UART sensors.
'
When I **ran 'dmesg | grep tty'** (a common command used to determine what UART ports there are) in the linux command line, I got:
'
robot@ev3dev:~$ dmesg | grep tty
[    0.000000] Kernel command line: console=tty1
[    0.002602] console [tty1] enabled
[    2.008769] **serial8250.0: ttyS0** at MMIO 0x1c42000 (irq = 25, base_baud = 9375000) is a TI DA8xx/66AK2x
[    2.012190] **serial8250.1: ttyS1** at MMIO 0x1d0c000 (irq = 53, base_baud = 8250000) is a TI DA8xx/66AK2x
[    2.015246] **serial8250.2: ttyS2** at MMIO 0x1d0d000 (irq = 61, base_baud = 8250000) is a TI DA8xx/66AK2x
[    3.173077] **1c30000.serial: ttySU0** at MMIO 0x1d00000 (irq = 254, base_baud = 8250000) is a suart_tty
[    3.180938] **1c30000.serial: ttySU1** at MMIO 0x1d00000 (irq = 256, base_baud = 8250000) is a suart_tty
[   20.356569] systemd[1]: Created slice system-getty.slice.
'
It listed only 5 which stated the word 'serial' in it's name. - ttyS0, ttyS1, ttyS2, ttySU0 and ttySU1. I don't know if it is a coincidence, but I feel like they corresponded to the ports listed in the webpage above, where,

**UART0 = ttyS0
UART1 = ttyS1
UART2 = ttyS2
PRU0/SUART0 = ttySU0
PRU0/SUART1 = ttySU1**

So I used **minicom** (a utility where you can send and receive bytes of data through a specific UART port eg. /dev/ttyS0 - the **baud rate** used was the default **115200** in case you were wondering) on both ev3 bricks and connected them with a cable and bluetooth in hopes of it working magically. **It didn't work**.

Question: How would I be able to use wire and bluetooth UART to communicate between 2 bricks? and what was I doing wrong above? (I am aware I might be able to use pyBluez)

Question (more important): How would I be able to interface between a sensor UART port on the ev3 brick and an arduino with a wire? (what UART ports to use on the ev3?)

I couldn't find any documentation about this online unfortunately, So thanks for the reply!

Version: ev3dev-stretch R1

dlech, 2019-04-25:
### Wired connection

You have the right idea. There are two more things you need to do. A) put the ports you are connecting into `other-uart` mode and B) create a crossover cable to connect the two EV3s.

A regular sensor/motor cable won't work because you will be connecting TX to TX and RX to RX. Also, the 5V lines from each brick could be fighting each other, so it is best to leave them disconnected. I think cutting a cable and rewiring it like this should work:

[image]

Also, if you are sending lots of data, you will probably need to lower the baud rate to avoid overflowing the hardware buffer. 9600 baud is a safe bet (and rather slow).

### Bluetooth

I would recommend searching for "bluetooth sockets linux" rather than trying to setup Bluetooth as a serial port.

runningOutOfUsernameIdeas, 2019-04-26:
Thanks for the reply!

I have got the wired UART connection working.

I am however still curious about 'UART Bluetooth'. Is there any reason why I can't use it / why is it not suggested? Is it, like, designed for something else?

Thanks.

dlech, 2019-04-26:
Bluetooth sockets use RFCOMM to communicate. I don't think it is any different than "UART Bluetooth" other than you use a socket instead of a tty. Unless you are trying to make an interactive terminal or something like that.

runningOutOfUsernameIdeas, 2019-05-05:
I understand now, thanks for the reply.

closed 2019-05-05 by runningOutOfUsernameIdeas