ev3dev/ev3dev #1268: Change tethering IP

xdsl, 2019-04-25:
How to change tethering IP 192.168.0.1 ?

dlech, 2019-04-25:
As far as I know, you can't. You could check the ConnMan mailing list to see if there have been any updates. It has been discussed in the past, but I don't think any action was ever taken.

xdsl, 2019-05-11:
Solved. I create udev rule for add tether network interface.

dlech, 2019-05-11:
Could you post your solution here in case others have the same problem?

xdsl, 2019-05-12:
ev3dev-jessie:
Create file /etc/udev/rules.d/99-bt-tether-net.rules:
`SUBSYSTEM=="net", ACTION=="add", DEVPATH=="/devices/virtual/net/tether", RUN+="/bin/sh -c 'ip addr add 192.168.1.1/24 dev tether'", RUN+="/bin/sh -c 'ip addr add 192.168.2.1/24 dev tether'"`

xdsl, 2019-05-12:
On computer I create to bluetooth ev3dev-connection in NetworkManager with static ip 192.168.1.2/24 or 192.168.2.2/24 and thus avoid conflicts with my home WIFI-network 192.168.0.0/24