ev3dev/ev3dev #1269: GyroSensor angle for GYRO-ANG sometimes returns empty string instead of integer

dwalton76, 2019-04-26:
**Describe the bug**
When reading `GYRO-ANG` sometimes we get `''` as the value, this causes the python library to barf because we are expecting an integer
```
Traceback (most recent call last):
  File "./utils/move_differential.py", line 55, in <module>
    mdiff.turn_to_angle(SpeedRPM(40), 90)
  File "/home/robot/ev3dev-lang-python/ev3dev2/motor.py", line 2442, in turn_to_angle
    self.turn_left(speed, angle_delta, brake, block)
  File "/home/robot/ev3dev-lang-python/ev3dev2/motor.py", line 2284, in turn_left
    self._turn(speed, abs(degrees) * -1, brake, block)
  File "/home/robot/ev3dev-lang-python/ev3dev2/motor.py", line 2208, in _turn
    gyro_init_angle = self.gyro.angle
  File "/home/robot/ev3dev-lang-python/ev3dev2/sensor/lego.py", line 597, in angle
    return self.value(0)
  File "/home/robot/ev3dev-lang-python/ev3dev2/sensor/__init__.py", line 219, in value
    self._value[n], value = self.get_attr_int(self._value[n], 'value'+str(n))
  File "/home/robot/ev3dev-lang-python/ev3dev2/__init__.py", line 291, in get_attr_int
    return attribute, int(value)
ValueError: invalid literal for int() with base 10: ''
```

**To Reproduce**
I haven't been able to come up with a set of bash commands yet that also triggers this but I can reproduce it pretty consistently.  I hit this while working on adding gyro support to be able to turn more accurately so you'll need to checkout a branch of mine to reproduce it.

1. Build [RileyRover](http://www.damienkee.com/rileyrover-ev3-classroom-robot-design/) but add a GyroSensor
2. git clone -b odometry-gyro https://github.com/dwalton76/ev3dev-lang-python ev3dev-lang-python-dwalton76
3. cd ev3dev-lang-python-dwalton76  # point PYTHONPATH to here
4. ./utils/move_differential.py

**Expected behavior**
To have it return the angle as an integer

**Robot (please complete the following information):**
RileyRover using an EV3

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165346201F
BOARD_INFO_TYPE=main
```

dlech, 2019-05-26:
Possibly related? #1083

dlech, 2019-11-23:
Is there a test case that doesn't involve building a robot?

dlech, 2019-11-24:
Also, it looks like you are using threads. I can see how this could happen if the gryo is being read from two different threads in Python.

thread 1: `.seek(0)`
thread 2: .`seek(0)`
thread 1: `.read()`  returns value
thread 2: `.read()` returns empty string

dwalton76, 2019-12-30:
@dlech I think you hit the nail on the head.  With the use of a lock I am no longer hitting the problem.  This is interesting though, it means folks must not be using threads very much with ev3dev-lang-python (at least not where multiple threads access the same ``Device`` object which is the base class for all motors and sensors)

closed 2019-12-30 by dwalton76