ev3dev/ev3dev #1270: Boot sequence doesn't even begin

JaumeI, 2019-04-26:
**What are you trying to do?**
I am trying to boot the EV3 with the image given by Lego Education. I flashed the SD card as instructions. Once flashed, I get asked if I want to format a secondary partition, I tried to format it in fat32, or not formatting it, and the result is always the same.

**What did you expect to happen**
The program should have started

**What actually happened**

The program didn't start. The brick is stuck with the red light on, In the screen I only see "Mindstorms" "Starting". I've waited for about 20 minutes for it to start.

**Desktop (please complete the following information):**
 - OS and version:  Window 10
 - Connection Type: microsd
 - Software: Etcher

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: Official Lego Education  Image

**Background Information**
Sorry if this isn't the place to ask, but I haven't found anywhere else. I wanted to try with my students the python programming, and got stuck.

BertLindeman, 2019-04-26:
Could it be that you should NOT format the sdcard after flashing?
The format overwrites you image, I think.

JaumeI, 2019-04-26:
Makes sense. I tried it already, giving it a go again. Maybe I need to format the rest of the SD card, where it doesn't have any partition? Or I better leave it unformatted? I mean the free space after the 2 partitions etcher created.

BertLindeman, 2019-04-26:
The first time you boot the new created image the main partition  is enlarged to fully fill the card.
That keeps the original image small so it can be burned onto smaller and onto larger sd-cards.

One question: is the size of the sd-card 32Gb or less? The EV3 **cannot** handle sd-card larger than 32Gb.

JaumeI, 2019-04-26:
It's a andisk Ultra 32GB microSDHC with A1 speed.
I'm updating the initial post with all this info. Thanks.

dlech, 2019-04-26:
What version of Etcher are you using? There is a bug in v1.5.30 (released 2 days ago) that causes corrupt images to be written. https://github.com/balena-io/etcher/issues/2754

You can download a working version of Etcher from https://github.com/balena-io/etcher/releases/tag/v1.5.26

JaumeI, 2019-04-26:
That was it! Awesome! Thanks!

closed 2019-04-26 by JaumeI

laurensvalk, 2019-04-30:
Hey @JaumeI,

The Etcher team just released an update with a fix proposed by @dlech. It's the latest one on the main page: https://www.balena.io/etcher/

Would you mind trying this, even though you got it to work already?

That would make it easier for us to help other users with the same issue. We would highly appreciate it :+1:

artpoz, 2019-04-30:
I checked the latest version of Etcher (1.5.23) and flashed the latest ev3dev (stretch-2003-03-03).
All works fine. You can update your 'Getting Started' page now.