ev3dev/ev3dev #1272: VS code hello python master not working

sugarfree78, 2019-04-29:
**What are you trying to do?**
I am trying run hello python master code through bluetooth on my EV3 brick. the status of the device is on line in Visual studio code compiler but the code is not running on brick. 

**What did you expect to happen**
the code should work through blue-tooth connection

**What actually happened**
the print code displays the message in the terminal not on ev3 screen or outB  large motor is not connected

The program did..
gives error that outB large motor is not connected

**Source Code**

[image]
[image]

`Here is the program I am trying to run.
from ev3dev2.motor import MoveSteering, OUTPUT_B, OUTPUT_C

motor_pair = MoveSteering(OUTPUT_B, OUTPUT_C)

for i in range(4):

    # Move robot forward for 3 seconds
    motor_pair.on_for_seconds(steering=0, speed=50, seconds=3)

    # Turn robot left 90 degrees (adjust rotations for your particular robot)
    motor_pair.on_for_rotations(steering=-100, speed=5, rotations=0.5)`
**Desktop (please complete the following information):**
 - OS and version: Window 10
 - Connection Type: Bluetooth
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2018-11-13

`

**Background Information**
I am working on self driving EV3 mind storm. during training samples collection, it takes image through IP webcam saves the image as numpy array along with the command on that specific picture in .txt file. it will train neural net work using that data and after that it will go in self drive mode. in self drive it suppose to communicate to EV3 through bluetooth to give commands.

dlech, 2019-05-26:
1. Please update to the most recent version of ev3dev
2. Is there actually a large motor connected to port B? (No loose wires?)

sugarfree78, 2019-06-10:
No, no loose wires. I actually realized that I need to send commands directly to ev3 but unable to do so.  can you help me in this regard. this is my first time working with ev3. thank you